mathieucarbou/ESPAsyncWebServer #7: :( many errors after this evening's update

bradbeshaw, 2024-01-27:
c:\Projects\Arduino\libraries\ESP_Async_WebServer\src\AsyncEventSource.cpp: In member function 'void AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*)':
c:\Projects\Arduino\libraries\ESP_Async_WebServer\src\AsyncEventSource.cpp:193:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  193 |       ets_printf(String(F("ERROR: Too many messages queued\n")).c_str());
      |       ^~~~~~~~~~
      |       vswprintf
c:\Projects\Arduino\libraries\ESP_Async_WebServer\src\AsyncWebSocket.cpp: In member function 'IPAddress AsyncWebSocketClient::remoteIP() const':
c:\Projects\Arduino\libraries\ESP_Async_WebServer\src\AsyncWebSocket.cpp:801:28: error: call of overloaded 'IPAddress(unsigned int)' is ambiguous
  801 |         return IPAddress(0U);
      |                            ^
In file included from C:\Users\brad\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\cores\esp32/Arduino.h:193,
                 from c:\Projects\Arduino\libraries\ESP_Async_WebServer\src\AsyncWebSocket.cpp:21:
C:\Users\brad\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\cores\esp32/IPAddress.h:51:5: note: candidate: 'IPAddress::IPAddress(const uint8_t*)'
   51 |     IPAddress(const uint8_t *address);
      |     ^~~~~~~~~

bradbeshaw, 2024-01-27:
is there a link to the prior version?

mathieucarbou, 2024-01-28:
Hello,
If you want help, please give as much info as possible.

closed 2024-03-07 by github-actions[bot]