mathieucarbou/ESPAsyncWebServer #11: Routing madness

trullock, 2024-02-20:
if you do

```
server.on("/foo", HTTP_GET, [] (AsyncWebServerRequest *req) {
    // A
});
server.on("/foo/bar", HTTP_GET, [] (AsyncWebServerRequest *req) {
    // B
});
```

And then make a request to "/foo/bar", you hit the first handler.

`AsyncCallbackWebHandler` and `AsyncCallbackJsonWebHandler` both have this in them:

```
if(_uri.length() && (_uri != request->url() && !request->url().startsWith(_uri+"/")))
      return false;
```

Which is the offending line. Why on earth is it doing a `startsWith`? 

We have regex and wildcard routing capabilities. I've not rummaged in Git history, is this a hangover from a previous, less powerful routing feature?

mathieucarbou, 2024-03-14:
This is how the original library work - you can look at its more complete documentation. 
This won't change in this fork, which is mainly there for bug fixes and maintenance purposes.

You need to place your most specific handlers first, and the most generic one after, when they both work on the same HTTP method.

My guess it to allow some features to work in the framework.
It also allow to be compliant with REST API standard with just a simple handler:

`GET /api/user/1234`
`PUT /api/user/1234`
etc

Where the handler would only be in `GET /api/user` and then the handler would read the user id from the URI.

closed 2024-03-14 by mathieucarbou

trullock, 2024-03-14:
Thanks for the reply