mathieucarbou/ESPAsyncWebServer #16: No way to determine whether an AsyncJsonResponse overflowed

tbnobody, 2024-03-31:
Currently it seems that there is no proper way to determine whether the amount of available memory was enough to store all the data in the `DynamicJsonDocument` or `JsonDocument` within `AsyncJsonResponse`. 
Both classes would provide a [overflowed](https://arduinojson.org/v6/api/jsondocument/overflowed/) method but the `_jsonBuffer` is marked private. It seems also not to be possible to get the parent object of the `_root` object to call the `overflowed` method.
Do you know any other method? Or could you add a `overflowed` method to the `AsyncJsonResponse` response class which just calls `_jsonbuffer.overflowed`?

mathieucarbou, 2024-04-01:
Good point. I ran into such issue in a project. There are other methods, like clear, that could be worth exposing because they only have an effect on the document itself.

The document itself could just be made available through a getter, which is returning a non const reference of the doc.

What do you think?

tbnobody, 2024-04-01:
aAmethod like `getDocument()` would indeed solve the problem. Question is, why this wasn't implemented in the beginning and instead the method `getRoot()` was implemented (already in the upstream project). 
Not sure if exposing the whole document object would lead to some bad side effects?

mathieucarbou, 2024-04-01:
The only reason I see is that if someone interacts with the document directly (i.e. clear), this will invalidate any other ref previously derived from a document, including _root... Which is bad ;-)

So yes, in that case, let's go with exposing a delegate for `overflowed()`