mathieucarbou/ESPAsyncWebServer #22: problème avec safari sur les fichiers compressé

xlyric, 2024-05-02:
Salut, 

Safari ne faisant rien comme les autres, lors qu'il charge une page, il peux charger une autre page dans certains cas 
exemple : 
---
 server.on("/log.html", HTTP_ANY, [](AsyncWebServerRequest *request){
        AsyncWebServerResponse *response = request->beginResponse(LittleFS, "/log.html.gz", "text/html");
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
  });
--- 

sur des navigateurs classiques, il prends log.html, sur safari il tentera de récuperer /log.html.gz

pour résoudre il faut virer la ligne 
544   addHeader(F("Content-Disposition"), buf);  dans WebResonses.cpp

mathieucarbou, 2024-05-02:
@xlyric : could you please test the fix by pointing to `https://github.com/mathieucarbou/ESPAsyncWebServer#issue-22` ?

Thanks :-)

xlyric, 2024-05-02:
je te tente de te fais ça dans la soirée 
cdt

xlyric, 2024-05-02:
c'est bon de mon coté, ça fonctionne. 
cdt
Cyril

mathieucarbou, 2024-05-02:
v2.9.5 released. Thanks!

PR #23 (2024-05-02): Remove filename after inline in Content-Disposition header according to RFC2183
Fix #22
closed 2024-05-02 by mathieucarbou