mathieucarbou/ESPAsyncWebServer #26: Captive Portal Filter AP Stopped Working on Arduino 3.0.0

rvbc1, 2024-05-29:
On this config:

```
[env]
platform = https://github.com/platformio/platform-espressif32.git#v6.7.0

platform_packages=
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0
  framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.0/esp32-arduino-libs-3.0.0.zip

board = esp32dev
framework = arduino

lib_deps = 
	mathieucarbou/ESP Async WebServer @ 2.10.0
        bblanchon/ArduinoJson @ 7.0.4
	
build_flags = 
	-std=gnu++17
	-DBOARD_HAS_PSRAM 
	-mfix-esp32-psram-cache-issue
	
build_unflags = 
	-std=gnu++11
	-std=gnu99
```

The AP filter for the captive portal example is not working:
`
server.addHandler(new CaptiveRequestHandler(requestHandleFunction, handleRequestBodyFunction)).setFilter(ON_AP_FILTER);  // only when requested from AP`

When I open the device web page (connected to the router, not directly to the ESP32), I also see the captive portal. On Arduino version 2.0.16 (latest from PlatformIO), everything works fine.

mathieucarbou, 2024-05-29:
@rvbc1 : can you please post your `.ino` modification to reproduce the issue. thanks!

mathieucarbou, 2024-05-29:
@rvbc1 : no need, i am able to reproduce. Looking...

mathieucarbou, 2024-05-29:
@rvbc1 : I've checked and the issue is caused by a commit from the ESPHome team when they introduced IPv6 support in AsyncTCP. I will fix the dependency.

mathieucarbou, 2024-05-29:
@rvbc1 : so I finally found the root cause of the issue. It comes from a bug in the new IPAddress implementation of Arduino 3: https://github.com/espressif/arduino-esp32/issues/9724

In the meantime, I will update the filter code for a workaround/

mathieucarbou, 2024-05-29:
I have implemented a workaround for https://github.com/espressif/arduino-esp32/issues/9724 in Async TCP 3.1.4. 

I am releasing v2.10.1 which will fix the bug, but ideally, the Arduino 3 team would need to fix the bug at the IPAddress class level.

About the ON_AP / ON_STA filters: I would not recommend using them (I myself do not use them). They rely on IP values comparison. If you have a use case where either the captive portal is on, either the website, then it is faster (and simple) to use `WiFi.getMode()` and check the returned enum.

The example is really more for the case where the ESP would be accessible from both STA and AP and you would need a different handler depending from which interface clients go through.

closed 2024-05-29 by mathieucarbou

vortigont, 2024-05-30:
BTW, in 3.0 they merged an updated DNS server based on AsyncUDP that I've PRed quite some time ago.
[Here](https://github.com/espressif/arduino-esp32/blob/master/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino) is how I implement captive portal there. It does not rely on Server's IP address filter but rather on HTTP client requesting unknown pages or foreign domain in `Host:`'s field.
Could be adopted to async server same way and should be more reliable to implement Captive portal.

mathieucarbou, 2024-05-30:
> BTW, in 3.0 they merged an updated DNS server based on AsyncUDP that I've PRed quite some time ago. [...]

That's the solution I use in my network manager lib: https://github.com/mathieucarbou/MycilaESPConnect ;-)

The example here in the project comes from the original forks, so I am leaving it as-is.

vortigont, 2024-05-30:
maybe leaving a reference link to Arduino core's example could be a good idea )