mathieucarbou/ESPAsyncWebServer #27: missing overload: void text(uint32_t id, std::shared_ptr<std::vector<uint8_t>> buffer);

tueddy, 2024-05-31:
The optimized websocket delivery works fine but I miss a possibility to notify a single client with optimized delivery. Can you maybe add an overlad function for`text()` same as `textAll()`?

### **void text(uint32_t id, std::shared_ptr<std::vector<uint8_t>> buffer);**

Current:
```
    void text(uint32_t id, const uint8_t * message, size_t len);
    void text(uint32_t id, const char *message, size_t len);
    void text(uint32_t id, const char *message);
    void text(uint32_t id, const String &message);
    void text(uint32_t id, const __FlashStringHelper *message);

```
Thanks & best regards
Dirk

mathieucarbou, 2024-05-31:
Of course! Will do it quickly in a few minutes ;-)

tueddy, 2024-05-31:
Do not rush ;-)

mathieucarbou, 2024-05-31:
Note that I strongly avise you to use the `AsyncWebSocketMessageBuffer` class instead, like in the spirit of the original fork. 

`std::shared_ptr<std::vector<uint8_t>>` was introduced in  yubox-node-org fork as an internal way of fixing the pointers scope / references, but this should be kept internal and not be dealt with. So I added back the AsyncWebSocketMessageBuffer which wraps that without overhead.

mathieucarbou, 2024-05-31:
`mathieucarbou/ESP Async WebServer @ 2.10.2` is released and in PlatformIO registry ;-)

closed 2024-05-31 by mathieucarbou

tueddy, 2024-06-02:
@mathieucarbou Thank's for the fast response & update! Everything is working now:
```
#if defined(ASYNCWEBSERVER_FORK_mathieucarbou)
	// serialize JSON in a more optimized way using a shared buffer
	const size_t len = measureJson(doc);
	auto buffer = std::make_shared<std::vector<uint8_t>>(len);
	serializeJson(doc, buffer->data(), len);
	if (client == 0) {
		ws.textAll(std::move(buffer));
	} else {
		ws.text(client, std::move(buffer));
	}
#else
..
```

I would agree to use a higher level thing like `AsyncWebSocketMessageBuffer ` but your example on the webpage shows implementation like above. Do you have a suggestion to rewrite code above for `AsyncWebSocketMessageBuffer ` and apply the example on the documentation page?
Please note if need booth, text() & textAll() without copying data around..

Thank's & best regards
Dirk

mathieucarbou, 2024-06-02:
This fork is based on youbox-node fork, which is based on the original repo so both documentation applies, to a certain extent because some features are not maintained.

I can add a note.

For your question, the doc is here:

https://github.com/me-no-dev/ESPAsyncWebServer?tab=readme-ov-file#direct-access-to-web-socket-message-buffer

But since I am mainly a maintainer but not the authors, I would appreciate any help and PR to keep this project alive 😊