mathieucarbou/ESPAsyncWebServer #47: Changes, required by using 3.0.2 version

Pablo2048, 2024-06-28:
Hi guys,
thanks for the hard work. This is no an issue, but info about what need to be done if one want to use latest 3.0.2 version in ESP8266 and ESP32 mixed environment.

  * as mentioned in doc some of parameters are now `const` (ie `AsyncWebParameter * p = request->getParam(i);` is now `const AsyncWebParameter * p = request->getParam(i);`
  * many `_P` methods for ESP32 are missing ie. `request->send_P(...` is now `request->send(...` but only for ESP32. For ESP8266 you have to still use `_P(...`
  * unfortunately to send array from progmem has now to be replaced by `beginResponse` ie `request->send_P(200, TEXTHTML, index_htm, index_htm_len);` is now `request->beginResponse(200, TEXTHTML, index_htm, index_htm_len);` but again only for the ESP32.

That is all I found out so far. Now I'm testing long term stability...

vortigont, 2024-06-28:
Hi @Pablo2048, are you using same code for both platforms in your project?
I mean for ESP32 you still have those `F("somethings")`?

mathieucarbou, 2024-06-28:
Hi @Pablo2048 : Infos in the release notes ;-)

https://github.com/mathieucarbou/ESPAsyncWebServer/releases

Don't forget that as per the semver convention each major digit increase is a breaking change, and you can control dependency lookup with for example `^3.0.2`.

FYI there is a pending PR that is pending review from @vortigont before I merge it that will put back the _P methods but with a deprecation notice to help ease the pain for people ;-)

mathieucarbou, 2024-06-28:
Moving to discussion as this is not an issue report.