mathieucarbou/ESPAsyncWebServer #52: [BUG] no matching function for call to 'AsyncWebServerRequest::beginResponse

Vano3d, 2024-06-30:
**Description**

Two days ago i got an compiling error "_no matching function for call to 'AsyncWebServerRequest::beginResponse_" right here:
AsyncWebServerResponse *response = request->**beginResponse**(200, "text/html", ELEGANT_HTML, sizeof(ELEGANT_HTML));

lib_deps = 
	https://github.com/me-no-dev/ESPAsyncWebServer.git
	me-no-dev/AsyncTCP @ ^1.1.1
	ayushsharma82/ElegantOTA@^3.1.2

build_flags=-DELEGANTOTA_USE_ASYNC_WEBSERVER=1 

**Board**

esp32dev

**Ethernet adapter used ?**
No

mathieucarbou, 2024-06-30:
Hello,

Use:

```
  mathieucarbou/Async TCP @ 3.1.4
  mathieucarbou/ESP Async WebServer @ 3.0.5
  https://github.com/ayushsharma82/ElegantOTA#3.1.2
```

You can look at the release notes for the 2 different projects. This is not a bug.

https://github.com/mathieucarbou/ESPAsyncWebServer/releases

closed 2024-06-30 by mathieucarbou

Vano3d, 2024-06-30:
Thank you a lot, it works!
One more thing: after replacing the library I had to add "**const**" before _AsyncWebParameter* p = request->getParam("data", true);_ in my handlers.

mathieucarbou, 2024-06-30:
Yes!
Again, in the release notes.
The api had been fixed in 3.x, should have been const since start, it was a mistake to consume the api without const. Now, const is enforced.