mathieucarbou/ESPAsyncWebServer #53: Can't reach the server in ESP32 CYD

AllanOricil, 2024-06-30:
**Please make sure to go through the recommendations before opening a bug report:**

[https://github.com/mathieucarbou/ESPAsyncWebServer?tab=readme-ov-file#important-recommendations](https://github.com/mathieucarbou/ESPAsyncWebServer?tab=readme-ov-file#important-recommendations)

**Description**

Server initialization throws no error, however I can't reach it in my local network at `{LOCAL_IP}/` or `{LOCAL_IP}/esp32`

[image]

I can ping the board from any device connected to the same network. In the image below you, can see that I can ping the board, however I can't curl `/`, which responds with a simple `Hello World`

[image]

This is the code I wrote to serve the files from the board

```c++
#include <Arduino.h>
#include <ESPAsyncWebServer.h>
#include <SPIFFS.h>
#include "constants.h"

bool checkFileExists(const char *path)
{
	if (SPIFFS.exists(path))
	{
		Serial.printf("File %s exists.\n", path);
		return true;
	}
	else
	{
		Serial.printf("File %s does not exist!\n", path);
		return false;
	}
}

void init_manager()
{
	Serial.println("Initializing manager server.");
	AsyncWebServer server(80);

	if (!SPIFFS.begin(true))
	{
		Serial.println("something went wrong while mounting SPIFFS");
		return;
	}

	if (
		!checkFileExists("/index.html") ||
		!checkFileExists("/404.html") ||
		!checkFileExists("/favicon.ico"))
	{
		Serial.println("One or more required files are missing. Initialization aborted.");
		return;
	}

	Serial.println("Configuring routes.");

	server.on(
		"/",
		HTTP_GET,
		[](AsyncWebServerRequest *request)
		{
			request->send(200, "text/plain", "Hello, world!");
		});

	server.on(
		"/esp32",
		HTTP_GET,
		[](AsyncWebServerRequest *request)
		{
			request->send(SPIFFS, "/index.html", "text/html");
		});

	server.onNotFound(
		[](AsyncWebServerRequest *request)
		{
			request->send(SPIFFS, "/404.html", "text/html");
		});

	server.serveStatic("/_nuxt/", SPIFFS, "/_nuxt/");

	server.on(
		"/favicon.ico",
		HTTP_GET,
		[](AsyncWebServerRequest *request)
		{
			request->send(SPIFFS, "/favicon.ico", "image/x-icon");
		});

	Serial.println("Routes configured.");

	server.begin();
	Serial.println("Server initialized.");
}
```

**Board**

ESP32-2432S028 (know as CYD)

**Ethernet adapter used ?**

WiFI

**Stack trace**

I'm not sure If it is necessary since there are no exceptions and the board is running with no problem. But if it is, please, guide me on how to dump the stack trace.

**Any issue opened with a non readable stack trace will be ignored because not helpful at all.**

As an alternative, you can use [https://maximeborges.github.io/esp-stacktrace-decoder/](https://maximeborges.github.io/esp-stacktrace-decoder/).

**Additional notes**

I tried 3 different forks and all got the same result. It is probably not an issue with the lib, but on my setup. However, I can't figure it out.

mathieucarbou, 2024-06-30:
Hello,

Please work on a [RME](https://en.wikipedia.org/wiki/Minimal_reproducible_example) that is not containing any specifics from your program, to determine if this is your code or the library.

If you can isolate a reproductible example in a .ino which still has the issue, file that you can paste, we can look at it.
In the meantime, I will flag this issue as invalid until it is proven valid.

Thanks!

mathieucarbou, 2024-06-30:
Also, juste FYI, `AsyncWebServer server(80);` needs to be at the top, not in a function, otherwise it will be destroyed at the end of the function. Please look at the many examples around to make sure your program is valid first.

AllanOricil, 2024-06-30:
@mathieucarbou your last commented worked. Thanks for seeing that mistake!

closed 2024-06-30 by AllanOricil