mathieucarbou/ESPAsyncWebServer #58: [Q]DEFAULT_MAX_SSE_CLIENTS

notixor, 2024-07-18:
DEFAULT_MAX_SSE_CLIENTS in AsyncEventSource.h not used anywhere

mathieucarbou, 2024-07-18:
You're totally right: this fork is based on:
- https://github.com/yubox-node-org/ESPAsyncWebServer
- https://github.com/esphome/ESPAsyncWebServer

`DEFAULT_MAX_SSE_CLIENTS` was added by ESPHome guys, and looking at it, even now, they just added it but they do not use it.

I think they wanted to implement a cleanup procedure like it is done in the websocket part.

Feel free to contribute and add this feature! 
Like said, since I do not have any SSE use case at hand, if there's no contributions, this part will not be improved.

mathieucarbou, 2024-12-21:
Closing - `DEFAULT_MAX_SSE_CLIENTS` was removed

closed 2024-12-21 by mathieucarbou