mathieucarbou/ESPAsyncWebServer #60: [BUG] literals.h compile error

andg1, 2024-07-18:
I'm getting a compiler error about reassignment in literals.h
Replacing 
#pragma once
with
#ifndef LITERALS_H
#define LITERALS_H
.
.
.
#endif

Fixes the error

mathieucarbou, 2024-07-18:
@vortigont : could you please have a look at this issue ? 

I suspect this is caused by compiler flags, but I think we should not use pragma for this project.

what do you think ?

mathieucarbou, 2024-07-18:
@andg1 : can you please show us the issue with the compilation flags ?

andg1, 2024-07-18:
I'm using Arduino IDE anyway I think those are the compiler flags

/home/andrea/.arduino15/packages/esp32/tools/esp-xs3/2302/bin/xtensa-esp32s3-elf-g++ -c @/home/andrea/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32s3/flags/cpp_flags -w -Os -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32S3_DEV" -DARDUINO_VARIANT="esp32s3" -DARDUINO_PARTITION_app3M_fat9M_16MB -DARDUINO_HOST_OS="linux" -DARDUINO_FQBN="esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=16M,PartitionScheme=app3M_fat9M_16MB,DebugLevel=none,PSRAM=opi,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 @/home/andrea/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32s3/flags/defines -I/home/andrea/

vortigont, 2024-07-18:
but what's the error actually?

andg1, 2024-07-18:
Compilation error: redefinition of 'constexpr const char* const asyncsrv::T_100_CONTINUE', is trowing errors for every assignment about esp32

mathieucarbou, 2024-07-18:
@andg1 : is it possible that you have a mix of libraries ? can you show us your .pio/libdeps folder if you use pio, or if using arduino ide, maybe cleanup your lib and redownload them ?

Because you see this line:

In file included from /.../ESPAsyncWebServer.h:46,

It is also surrounded with `#ifndef _ESPAsyncWebServer_H_`

vortigont, 2024-07-19:
agreed with [mathieucarbou](https://github.com/mathieucarbou), pls recheck if by any chance you have two instances of `literals.h` files somewhere under `.pio` or other include dirs

also note this:
`/.../literals.h:10:30: note: 'constexpr const char* const asyncsrv::T_100_CONTINUE' `
error message mentions line 10, while in current instance of `literals.h`  `T_100_CONTINUE` is defined in line 6.

andg1, 2024-07-19:
The line is different because I've commented out the ifdef lines and added back pragma, also in a different computer I'm getting the same error on arduino ide. Will fix it in my side if no other user is getting the same error. Thank you anyway

mathieucarbou, 2024-07-19:
> The line is different because I've commented out the ifdef lines and added back pragma, also in a different computer I'm getting the same error on arduino ide. Will fix it in my side if no other user is getting the same error. Thank you anyway

You can try to use PlatformIO, and also you can try to compile an exemple in the project, using platformuio, or Arduino IDE, but first cleanup your dependency folder and start fresh. It really looks like an IDE / env issue. Platformio is correctly handling dependencies, not like Arduino IDE.