mathieucarbou/ESPAsyncWebServer #63: [Q]I need help creating a websocket server with this library

sensslen, 2024-07-23:
I'm trying to create a little websocket server to control a video camera. Most things work flawlessly, however once I try to connect to the websocket, I'm always getting a 500 response....

I'm using postman to test the connection and this is the result I get:
```
Error: Unexpected server response: 500
Handshake Details
Request URL: http://192.168.178.64/ws/
Request Method: GET
Status Code: 500 Internal Server Error
Request Headers
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: b4AwU2XaRoThHnvM2WtP5Q==
Connection: Upgrade
Upgrade: websocket
Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
Host: 192.168.178.64
Response Headers
Content-Length: 0
Connection: close
Accept-Ranges: none
```

The code that runs on the ESP32 can be found [here](https://github.com/sensslen/Cgf.CameraControl.Camera). As you can tell I tried WiFi and Ethernet. Both do obtain an IP address from the network and respond to ping messages.

Any help is greatly appreciated!

mathieucarbou, 2024-07-23:
Hello,
You would need to be more specific with the symptoms.
If you add seevral prints when connecting to the WS handler (here: https://github.com/sensslen/Cgf.CameraControl.Camera/blob/main/src/WebsocketReceiver/WebSocketReceiver.cpp#L48-L59), you do not see them ?
Did you try connecting with websocat ?

sensslen, 2024-07-24:
> Hello,
>
> You would need to be more specific with the symptoms.
>
> If you add seevral prints when connecting to the WS handler (here: [...]

I'm sorry - no, this method does never get hit...

vortigont, 2024-07-24:
Try Request URL: `http://192.168.178.64/ws` (without trailing backslash)

sensslen, 2024-07-24:
This dos not change anything:

```
Could not connect to ws://192.168.178.64/ws
08:44:34
Error: Unexpected server response: 500
Handshake Details
Request URL: http://192.168.178.64/ws
Request Method: GET
Status Code: 500 Internal Server Error
Request Headers
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: nfvaRzqO51zAOKxTQUK6Og==
Connection: Upgrade
Upgrade: websocket
Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
Host: 192.168.178.64
Response Headers
Content-Length: 0
Connection: close
Accept-Ranges: none
```

sensslen, 2024-07-24:
Using websocat leads to the same response:
```
PS ***> .\websocat.x86_64-pc-windows-gnu.exe ws://192.168.178.64/ws/
websocat: WebSocketError: WebSocketError: Received unexpected status code (500 Internal Server Error)
websocat: error running
PS ***>
```

sensslen, 2024-07-24:
Is it possible this has something to do with the fact that I use the experimental arduino version 3 integration? https://github.com/sensslen/Cgf.CameraControl.Camera/blob/2bfa5b64cbff7e6eff9f52a43e03fde59af2aec7/platformio.ini#L18

mathieucarbou, 2024-07-24:
`ws://192.168.178.64/ws` not `ws://192.168.178.64/ws/`. Your handler is defined at `ws://192.168.178.64/ws`

mathieucarbou, 2024-07-24:
> Is it possible this has something to do with the fact that I use the experimental arduino version 3 integration? https://github.com/sensslen/Cgf.CameraControl.Camera/blob/2bfa5b64cbff7e6eff9f52a43e03fde59af2aec7/platformio.ini#L18

no not at all. there is something in your code...

sensslen, 2024-07-24:
OK, I realized that as well - switched to the currently supported arduino sdk and still see the issue. By the way:
```
PS ***> .\websocat.x86_64-pc-windows-gnu.exe ws://192.168.178.64/ws
websocat: WebSocketError: WebSocketError: Received unexpected status code (500 Internal Server Error)
websocat: error running
PS ***>
```

sensslen, 2024-07-24:
Do I have to wait for the ethernet/wifi connection to be established before I start the webserver?

mathieucarbou, 2024-07-24:
> Do I have to wait for the ethernet/wifi connection to be established before I start the webserver?

You need to begin the webserver once WiFi is connected, yes, but the websocket handler  does not have to be "started" - this is a handler.

Also, please have a look at: https://github.com/mathieucarbou/ESPAsyncWebServer?tab=readme-ov-file#coordinate-and-dependencies

you are using an old version of the lib.

sensslen, 2024-07-24:
Well then I guess I know why things are not working - I'm not waiting for the connection to be established before starting the server. Will adjust. Thank you so much for your assistance! It's really appreciated....

sensslen, 2024-07-24:
I'm now fully communicating. The issue was that the webserver was started before the network connection was established. Thanks again for pointing me to the right direction and also for the great library!

closed 2024-07-24 by sensslen