mathieucarbou/ESPAsyncWebServer #66: [Q]call of overloaded 'getParam(int)' is ambiguous

maxelle, 2024-07-29:
I receice this error comipiling

C:\Arduino2\test\test.ino:647:47: error: call of overloaded 'getParam(int)' is ambiguous
  647 |       AsyncWebParameter* p = request->getParam(0);

Please help
Thanks Max

maxelle, 2024-07-29:
Thanks for answer,
I tryed bt same error... 
[image]

mathieucarbou, 2024-07-29:
@maxelle : your parameter type is wrong. Please look at the header file. This is the doc.

maxelle, 2024-07-29:
so i have to cast like...
const AsyncWebParameter* p = request->getParam((size_t)0);

mathieucarbou, 2024-07-29:
> so i have to cast like... const AsyncWebParameter* p = request->getParam((size_t)0);

This is one way to do it, yes. `0` as written here is an `int`. There is no `getParam()` accepting `int`. And the compiler cannot know which one to pick.

closed 2024-07-29 by mathieucarbou