mathieucarbou/ESPAsyncWebServer #73: [Q]ESPAsyncWebServer.h question

lidodvr224, 2024-08-12:
Running newest versions, Windows/ArduinoIDE 232+ 30Jul24 nightly. (v313)
line157 change error:

Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h: In member function 'String AsyncWebHeader::toString() const':
c:\Users\jelli\Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:157:44: error: ambiguous overload for 'operator+' (operand types are 'const String' and 'char')
  157 |     String toString() const { return _name + (char)0x3a + (char)0x20 /*": "*/ + _value + asyncsrv::T_rn; }
      |                                      ~~~~~ ^ ~~~~~~~~~~
      |                                      |       |
      |                                      |       char
      |                                      const String

Thank you for any feedback.

mathieucarbou, 2024-08-12:
That's a bug.
@vortigont : FYI

mathieucarbou, 2024-08-12:
@lidodvr224 : can you please try with branch to verify this is fixed ?

 #bugfix (https://github.com/mathieucarbou/ESPAsyncWebServer/tree/bugfix) ?
 
 Thanks!

lidodvr224, 2024-08-12:
yes, standby...

lidodvr224, 2024-08-12:
v 3.13 with new ESPAsyncWebServer.h (28261 bytes) from 12Aug24 built, and
my application is working as expected with MycilaWebSerial (630).
will update this on github.
thank you!

PR #74 (2024-08-12): Fix #73
- Fix #73
closed 2024-08-12 by mathieucarbou