mathieucarbou/ESPAsyncWebServer #76: I am still facing this known bug: [BUG]https://github.com/me-no-dev/ESPAsyncWebServer/issues/900

VEP-DEV, 2024-08-22:
**Please make sure to go through the recommendations before opening a bug report:**

[https://github.com/mathieucarbou/ESPAsyncWebServer?tab=readme-ov-file#important-recommendations](https://github.com/mathieucarbou/ESPAsyncWebServer?tab=readme-ov-file#important-recommendations)

**Description**

A clear and concise description of what the bug is.

**Board**

esp32dev, esp32s3, etc

**Ethernet adapter used ?**

If yes, please specify which one

**Stack trace**

Please provide the stack trace here taken with `monitor_filters = esp32_exception_decoder`.
**Any issue opened with a non readable stack trace will be ignored because not helpful at all.**

As an alternative, you can use [https://maximeborges.github.io/esp-stacktrace-decoder/](https://maximeborges.github.io/esp-stacktrace-decoder/).

**Additional notes**

Add any other context about the problem here.

closed 2024-08-22 by mathieucarbou

VEP-DEV, 2024-08-23:
> @VEP-DEV <https://github.com/VEP-DEV> : this fork is based on yubox -node
> fork which fixes the issue you are referring to supposedly.
> Also this fork has diverged a lot.
> So please start over like a new issue.
>
> We need a minimal reproductible use case and some analysis from you, like
> a stack trace and eventually some memory (heap information) of memory
> related.
>
> Thanks.

Hi Mathieu,

This is a simple IoT device that collects sensor data and forwards them
using LTE modem to a server .
It is based on ESP32 and my development environment is Msvc + Platformio. I
use Arduino framework.
The device has a web interface using ESP32 as a hotspot so the user can
change some parameters through WiFi connection.
The problem is that the device restarts randomly because webserver
sometimes crashes.
As it happens rarely and randomly it is difficult to catch by logging the
Serial Monitor output. So I copy the backtrace data to a place of the RAM
that doesn't initialised after the soft reset and after the reset I save
them (in a json file) to a built-in SD card. Unfortunately saving heap info
doesn't work yet.

This is what I got:
{"Year":24,"Month":8,"Day":20,"Hour":8,"Minute":4,"Second":50,"Boot
reason":4,"heap_total":0,"heap_free":0,"heap_free_min":0,"heap_min_time":0,"backtrace0":"0x400D68AF","backtrace1":"0x400F7237","backtrace2":"0x40083675","backtrace3":"0x4008513C","backtrace4":"0x400DEBF8","backtrace5":"0x400DF51F","backtrace6":"0x400DF537","backtrace7":"0x400DF555","backtrace8":null}

And this is the decoded data:
d:\ZMonitorV4\SW\ZMonitor\zmonitor\Panic>xtensa-esp32-elf-addr2line.exe
-pfiaC -e firmware.elf 0x400D68AF 0x400F7237 0x40083675 0x4008513C
0x400DEBF8 0x400DF51F 0x400DF537 0x400DF555
0x400d68af: __wrap_esp_panic_handler at
C:\B+N\ZMonitorV4\SW\Munka\zmonitor/src/ZMonitor.cpp:1567
0x400f7237: panic_handler at
/home/runner/work/esp32-arduino-1ib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/panic_handler.c:188
0x40083675: xt_unhandled_exception at /home/runner
/work/esp32-arduino-lib-builder
/esp32-arduino-1ib-builder/esp-idf/components/esp_system/port/panic_handler.c:219
0x4008513c: _xt_user_exc at
/home/runner/work/esp32-arduino-1ib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:703
0x400debf8: AsyncWebSocket: :_cleanBuffers() at
C:\B+N\ZMonitorV4\SW\Munka\zmonitor/.pio/libdeps/zmon/ESP Async
WebServer/src/AsyncWebSocket.cpp:1229 (discriminator 2)
0x400df51f: AsyncWebSocket: :textAll(AsyncWebSocketMessageBuffer*) at
C:\B+N\ZMonitorV4\SW\Munka\zmonitor/.pio/libdeps/zmon/ESP Async
WebServer/src/AsynchlebSocket.cpp:962
0x400df537: AsyncWebSocket: :textAll(char const*, unsigned int) at
C:\B+N\ZMonitorV4\sW\Munka\zmonitor/.pio/libdeps/zmon/ESP Async
WebServer/src/AsyncebSocket.cpp:968
0x400df555: AsyncWebSocket: :textAll(String const&) at
C:\B+N\ZMonitorV4\SW\Munka\zmonitor/.pio/libdeps/zmon/ESP Async
WebServer/src/AsyncWebSocket.cpp:1109

I updated the ESPAsyncWebServer, AsyncTCP and ArduinoJson but the problem
still exists.
Thank you for your help in advance!

Best regards,
Peter

Mathieu Carbou ***@***.***> ezt írta (időpont: 2024. aug.
22., Cs, 18:30):

mathieucarbou, 2024-08-23:
Would you be able to log and get the complete stack trace with the register values ?

mathieucarbou, 2024-08-23:
> 0x400debf8: AsyncWebSocket: :_cleanBuffers() at C:\B+N\ZMonitorV4\SW\Munka\zmonitor/.pio/libdeps/zmon/ESP Async WebServer/src/AsyncWebSocket.cpp:1229 (discriminator 2)

Also please make sure to use the right librairie : the stack trace does not match the source code: please have a look at the source code lines first to be sure it matches.

VEP-DEV, 2024-08-23:
The trace I sent comes from an earlier phase when I used the me-no-dev fork.

Complete tracking info I can get only by using logging data to the PC using
Serial Monitor and EspProg tool.
It take some time when the crashing occurs. I don't know how long that is.
So I switch the device on, logging the data and wait, wait, wait,
..............
I'll contact you when I catch the issue.

Mathieu Carbou ***@***.***> ezt írta (időpont: 2024. aug.
23., P, 12:26):

mathieucarbou, 2024-08-23:
So please open this issue in the right fork. I will keep this one closed.
If you have to see an issue when using this fork , then open an issue with the stack trace and register state marching the code.
Thank you.

VEP-DEV, 2024-08-23:
Ok, I see.
But I would like to emphasize that the crash occured in this fork also.
I am working to catch this one also.
Thank you for your support.

Mathieu Carbou ***@***.***> ezt írta (időpont: 2024. aug.
23., P, 13:04):

mathieucarbou, 2024-08-23:
> But I would like to emphasize that the crash occured in this fork also.

That's exactly what is unclear. 
As of now, you have just produced an issue that is completely unrelated to this fork: the source code does not match. 
There are so many big projects using this fork with heavy websocket usage that I find it strange. If such issue existed, it would have been caught I think.
Also, the issue you are referring to is supposed to be fixed in the fork

> I am working to catch this one also.

Thanks! Please open a new issue if you can catch it again with this fork and you have a complete stack trace that can be analysed.