mathieucarbou/ESPAsyncWebServer #77: [Feat] Allow modification of hardcoded headers (Accepts Range, Content-Length)

DRSDavidSoft, 2024-08-29:
### Description

Hi there, in order to support a custom file serving scenario (from UART) that also supports `HEAD` request to be able to resume downloads, I need to override two hard coded headers:

```cpp
// ...
response->acceptsRange(true);
response->sendContentLength(false);
```

Now it would be awesome if you could implement it, so that instead of using a _fork_ of a fork of a fork, this feature would exist in this repo :)

```
String AsyncWebServerResponse::_assembleHead(uint8_t version){
  if(version){
    if(!_acceptsRange)
      addHeader(F("Accept-Ranges"), F("none"));
    if(_chunked)
      addHeader(F("Transfer-Encoding"), F("chunked"));
  }
```

I can also send a PR if you will accept this. Thank you for maintaining ESPAsyncWebServer!

### Board

ESP8266

mathieucarbou, 2024-08-30:
Hi,
Yes please send a PR, @vortigont and/or me will review for sure!
Do not forget to put the strings in the bundle made for them please ;-)

DRSDavidSoft, 2024-08-30:
Sure! I just noticed the string bundles.

DRSDavidSoft, 2024-08-30:
I submitted the PR, this compiles but I haven't tested it out yet - the code matches with the patch I'm using in my project, so it should be fine.

Once again thanks for maintaining this awesome library! 👍🏻

mathieucarbou, 2024-08-31:
@DRSDavidSoft : could you please have a look at PR https://github.com/mathieucarbou/ESPAsyncWebServer/pull/82 and test it ? I think it should be close to what you need and allows the user to globally control headers override.

commit 85de3863 (2024-09-01): Close #77: Allow modification of hardcoded headers
commit f0402e5a (2024-09-01): Close #77: Allow modification of hardcoded headers
closed 2024-09-01 by mathieucarbou