mathieucarbou/ESPAsyncWebServer #86: Fix potential duplicate of content-type and content-length

mathieucarbou, 2024-09-02:
>  currently setting a custom Content-Type using addHeader() before _assembleHead() will result in duplicate headers.

@DRSDavidSoft : oh you are right I didn't pay attention to that...

This is the library provides another way to set he content-type and content-length... I guess these headers should be intercepted maybe....

> Which is why I ultimately decided against modifying the original addHeader() method.

Damn! You are right! I didn't pay attention to this use case. I will add a breaking change warning. I think this is acceptable.

_Originally posted by @mathieucarbou in https://github.com/mathieucarbou/ESPAsyncWebServer/issues/82#issuecomment-2324914509_

DRSDavidSoft, 2024-09-02:
> the library provides another way to set he content-type and content-length

This is a non-issue, the library will keep track of these core headers internally and will insert them into the response header at the last moment where everything else is handled.

If the user decides to also call `addHeader()` with the `Content-Type` or `Content-Length` headers, they're essentially _overriding_ these "headers", not the value of `_contentType` and `_contentLength` properties, per se.  I hope I'm getting my point across.

In this case, simply doing something like this is a good solution in my opinion, because if the inherit `_contentType` and/or `_contentLength` are set, those will be added when `_assembleHead()` is called, unless the user has _already_ specified a header using `addHeader()`:

```
snprintf_P(buf, bufSize, PSTR("Content-Length: %d\r\n"), _contentLength);
// → addHeader(T_Content_Length, String(_contentLength), false)

snprintf_P(buf, bufSize, PSTR("Content-Type: %s\r\n"), _contentType.c_str());
// → addHeader(T_Content_Type, _contentType, false)
```

> I guess these headers should be intercepted maybe

Alternative approach would be this, which is also a great way to do this, you will be essentially comparing the header names when the user calls `addHeader()`. Then if the header name matches, you will insert the value of the `Content-Type` into the `_contentType` variable. This has the added benefit that it can also later be accessed with `getContentType()`if such a method is added.

For the `Content-Length` header, it's pretty much the same, however you will also need to parse the value of the header, from string to an integer. If you are going to do this, I will also suggest adding a `addHeader(..., (long))` override so that the user could do this:

```
addHeader("Content-Type", 12345);
```

instead of:

```
addHeader("Content-Type", String(12345));
```

And lastly, the original `setContentType()` and `setContentLength()` must still be kept, both because of backwards compatibility and also because these are core HTTP headers that define how the library behaves (more importantly the content length one).

DRSDavidSoft, 2024-09-02:
Also, sorry if it's taking so much development resources, ideally such modifications should be [KISS](https://en.wikipedia.org/wiki/KISS_principle), but I'm really invested in applying a fix that is both backwards compatible and also a comprehensive way to go about implementing it.

mathieucarbou, 2024-09-02:
thanks for the summary!

Do you want to open a PR and update the example ? I agree with your proposal. Let's reuse what is in place.
This PR should be quite strait forward ;-)

Otherwise I will do in a a few hours, later.

DRSDavidSoft, 2024-09-02:
If it's alright, I'll wait to see how you implement it, I appreciate the effort!

mathieucarbou, 2024-09-02:
@DRSDavidSoft : I have implemented this PR and tested using the example and curl - seems to work.

https://github.com/mathieucarbou/ESPAsyncWebServer/pull/88

I went a little further by also updating setContentLength() and setContentType() methods to sync the calls with the headers 

This allows users to either call these 2 or addHeader() to set and override the content-type / length

Would you be able to test before I issue a release (either the PR branch, or master once I will have merged) ?

Thanks!

DRSDavidSoft, 2024-09-02:
@mathieucarbou Sure, I'll test in a few hours and will let you know, thank you for the update!

DRSDavidSoft, 2024-09-03:
@mathieucarbou This is working wonderfully, the file is being downloaded correctly and without any issues! 🎉 

The headers are all being sent correctly! And thanks for the addition of new methods, it's very pleasing to do `addHeader("Content-Type", myFile.getSize())` instead of `addHeader("Content-Type", String(myFile.getSize()))` 😄 

I see that the new PR also addresses the String usages in the old library, you are minimalizing the need on dynamic allocation and also the `printf()`s. This is great performance wise, despite reducing code readability just a tiny bit. 👍🏻 

### One more final thing, for the sake of completeness
There is one use case where we _don't want_ to send the content type and content length as it's unknown. We can just send an empty content length, and `_contentType.length()` will be zero, which means the Content-Type header will not be set during `_assembleHead()`. ✅

Now, what if the content length is also unknown? Passing zero as `setContentLength()` will result in setting `Content-Length: 0` which is not the desired effect in this case.

Please note that this is different the Content-Length _actually_ being zero, which would be the correct result of `setContentLength(0)`.

With this being said, I propose that a new method is added (`sendContentLength()`) that can be set to false by the user, thus allowing control over this last piece of the puzzle.

As always, thank you so much for this great quality of code, I'm amazed by the excellent approaches you take to implement these, truly exciting stuff! 😄 👍🏻

mathieucarbou, 2024-09-03:
Yes, `_sendContentLength ` is controlled and set to true only in `AsyncFileResponse`. I will issue a new release! thanks for the feedback!

commit ce563bf5 (2024-09-02): Fix header handling versus internal handling of content-type and content-length (Fix #86)
commit f9cfba33 (2024-09-02): Fix header handling versus internal handling of content-type and content-length (Fix #86)
PR #88 (2024-09-03): Fix header handling versus internal handling of content-type and content-length
Fix #86
closed 2024-09-03 by mathieucarbou