mathieucarbou/ESPAsyncWebServer #87: [BUG] beginResponse_P works - beginResponse - crashing

DaeMonSxy, 2024-09-02:
**beginResponse_P works okay, but it will be deprecated, 
so i turned to beginResponse ... now crashing..**

platform: esp8266

page content:
```
const uint8_t bootstrap_css[] PROGMEM = {
0x1f,0x8b,0x08,......

```

```
void sendStaticResponse(AsyncWebServerRequest *request, const uint8_t *htmlContent, size_t htmlSize, const char *contentType)
{
    auto response = request->beginResponse_P(200, contentType, htmlContent, htmlSize);
    response->addHeader(FPSTRP(CONTENT_ENCODING), FPSTRP(GZIP));
    request->send(response);
}

WServer.on("/bootstrap.css", HTTP_GET, [](AsyncWebServerRequest *request) {
        sendStaticResponse(request, bootstrap_css, sizeof(bootstrap_css), FPSTRP(TEXT_CSS));
    });
```

[[deprecated("Replaced by beginResponse(...)")]]
    AsyncWebServerResponse* beginResponse_P(int code, const String& contentType, const uint8_t* content, size_t len, AwsTemplateProcessor callback = nullptr) {
      return beginResponse(code, contentType, content, len, callback);
    }

**crash:**
> 0x4026688f in system_get_sdk_version at ??:?
0x4021dd19 in AsyncAbstractResponse::AsyncAbstractResponse(std::function<String (String const&)>) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebResponses.cpp:405
0x402685c4 in system_get_sdk_version at ??:?
0x4026688f in system_get_sdk_version at ??:?
0x4021dda5 in AsyncProgmemResponse::AsyncProgmemResponse(int, char const*, unsigned char const*, unsigned int, std::function<String (String const&)>) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebResponses.cpp:826
0x4026688f in system_get_sdk_version at ??:?
0x40100c18 in malloc at /Users/daemons/platformio_packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:912
0x402685c4 in system_get_sdk_version at ??:?
0x4026688f in system_get_sdk_version at ??:?
0x4021b1cc in AsyncWebServerRequest::beginResponse(int, char const*, unsigned char const*, unsigned int, std::function<String (String const&)>) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebRequest.cpp:699
0x4021b832 in AsyncWebServerRequest::addInterestingHeader(char const*) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebRequest.cpp:690
0x402685c4 in system_get_sdk_version at ??:?
0x4026688f in system_get_sdk_version at ??:?
0x40208c26 in sendHTMLResponse(AsyncWebServerRequest*, unsigned char const*, unsigned int, char const*, html_pages_t) at ./web_handlers/WServer.h:66
0x4021a457 in _ZZN21AsyncWebServerRequest28_removeNotInterestingHeadersEvENKUlRK6StringE_clES2_$constprop$0 at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebRequest.cpp:156
0x4021ea78 in AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at /Users/daemons/platformio_packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:622
0x4021eeac in AsyncWebServer::_attachHandler(AsyncWebServerRequest*) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebServer.cpp:152
0x4021c914 in AsyncWebServerRequest::_parseLine() at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebRequest.cpp:587
0x4021ca7c in AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebRequest.cpp:94
0x40102f40 in rcReachRetryLimit at ??:?
0x401033fa in wDev_ProcessFiq at ??:?
0x402166e0 in ACErrorTracker::hasClient() const at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncTCP-esphome/src/ESPAsyncTCP.h:103
 (inlined by) AsyncClient::_recv(std::shared_ptr<ACErrorTracker>&, tcp_pcb*, pbuf*, int) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncTCP-esphome/src/ESPAsyncTCP.cpp:649
0x40216748 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, int) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncTCP-esphome/src/ESPAsyncTCP.cpp:740
0x4023c099 in tcp_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c:501 (discriminator 1)
0x40241001 in ip4_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1467
0x40238b49 in mem_malloc at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:210
0x40100896 in umm_free_core at /Users/daemons/platformio_packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:642
0x4025ded0 in ppRecycleRxPkt at ??:?
0x4023820d in ethernet_input_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:188
0x4023801c in git2glue_err at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:118
 (inlined by) esp2glue_ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:494
0x40261095 in ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:365
0x402610a7 in ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:373
0x4025daff in ppPeocessRxPktHdr at ??:?
0x40262383 in ets_snprintf at ??:?
0x40105571 in call_user_start_local at ??:?
0x40105577 in call_user_start_local at ??:?
0x4010000d in call_user_start at ??:?
[... 70 lines cut ...]
0x40104364 in lmacProcessTXStartData at ??:?
0x401032cb in wDev_ProcessFiq at ??:?
0x4010311c in wDev_ProcessFiq at ??:?
0x4010311c in wDev_ProcessFiq at ??:?
0x4022f339 in _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:193 (discriminator 1)
0x401002a8 in ets_post at /Users/daemons/platformio_packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:238
0x40104fa2 in lmacMSDUAged at ??:?
0x401021d8 in pp_post at ??:?
0x4025dd63 in ppTxPkt at ??:?
0x401048f1 in lmacRecycleMPDU at ??:?
0x40246fb3 in ieee80211_output_pbuf at ??:?
0x40104ea5 in lmacMSDUAged at ??:?
0x40233efc in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x401053bd in lmacTxFrame at ??:?
0x401002a8 in ets_post at /Users/daemons/platformio_packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:238
0x40102609 in rcUpdateTxDone at ??:?
0x401021d8 in pp_post at ??:?
0x4010546f in lmacRxDone at ??:?
0x40102d63 in rcReachRetryLimit at ??:?
0x4022f494 in _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:246
0x40102f40 in rcReachRetryLimit at ??:?
0x401033fa in wDev_ProcessFiq at ??:?
0x402340f9 in _svfprintf_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:528
0x4010546f in lmacRxDone at ??:?
0x40102d63 in rcReachRetryLimit at ??:?

DaeMonSxy, 2024-09-02:
Well can be...
However I am not loading pages one after another in fast pace.

After esp8266 bootup, nothing happens ( no crash)
then if i type in the webpage, immediatelly crashing - no reloads or fast clicks, not intentionall communication (ws)

Hence I did not link that to "fast page change" - thats heap issue i believe

mathieucarbou, 2024-09-02:
@DaeMonSxy : if you can adapt one of the example in the project, or send me an .ino file I can put in the project (in examples) as a reproductible use case, I can try to have a look with my D1 Mini.

lumapu, 2024-09-02:
can you tell us what is behind the macro `FPSTRP`.
Have you tried it without the macro?

```cpp
AsyncWebServerResponse *response = request->beginResponse(200, F("text/html; charset=UTF-8"), zippedHtml, len);
response->addHeader(F("content-type"), "text/html; charset=UTF-8");
response->addHeader(F("Content-Encoding"), "gzip");
request->send(response);
```

I too use ESP8266 with latest library version, serving g-zipped contents without any problem.

mathieucarbou, 2024-09-02:
Thanks @lumapu ! I didn't even notice FPSTR`P` ! ;-)

DaeMonSxy, 2024-09-03:
yes, i have tried "simplified" versioned also.

```
// Using F() macro to store strings in flash memory
const char TEXT_HTML[] PROGMEM        = "text/html";
const char TEXT_CSS[] PROGMEM         = "text/css";
const char APPLICATION_JS[] PROGMEM   = "application/javascript";
const char CONTENT_ENCODING[] PROGMEM = "Content-Encoding";
const char GZIP[] PROGMEM             = "gzip";

#define PSTRP(s)  (reinterpret_cast<const __FlashStringHelper *>(s))
#define FPSTRP(s) (reinterpret_cast<const char *>(s))

```

DaeMonSxy, 2024-09-03:
> @DaeMonSxy : if you can adapt one of the example in the project, or send me an .ino file I can put in the project (in examples) as a reproductible use case, I can try to have a look with my D1 Mini.

well, im not sure i can do that soon, going to be busy this days.

I do not use ino or arduino in past few years - im on platformio.
having a quite big code, but will check closer to weekend if i can streamline the code just to have this block of code extracted for checking.

meanwhile,  did a bit of checking, if i  pass html code: - it is working OK
**`const char SAMPLE_HTML[] PROGMEM      = "<html><body><h1>Sample Page</h1></body></html>";`**

the crash occurs if used:  uint  - crashing
```
**const uint8_t system_html[] PROGMEM = {
0x1f,0x8b,....
};**

```

mathieucarbou, 2024-09-03:
> meanwhile, did a bit of checking, if i pass html code: - it is working OK

Yeah but it prevents you from gzip encoding, so not ideal...

Did you try same code as @lumapu, without using your macros ?

DaeMonSxy, 2024-09-03:
**yes, same - crashing :**
same without progmem variables below.

`
const char CONTENT_ENCODING[] PROGMEM = "Content-Encoding";
const char GZIP[] PROGMEM             = "gzip";

const uint8_t system_html[] PROGMEM = {
0x1f,0x8b,.......
};

//----------------------------------------------------------------------------

void sendHTMLResponse(AsyncWebServerRequest *request, const uint8_t *htmlContent, size_t htmlSize, const char *contentType, html_pages_t setpage)
{
      auto response = request->beginResponse(200, contentType, htmlContent, htmlSize);
    response->addHeader(CONTENT_ENCODING, GZIP);
    request->send(response);
    current_html_page = setpage;
}

WServer.on("/syst", HTTP_GET, [](AsyncWebServerRequest *request) {
        sendHTMLResponse(request, system_html, sizeof(system_html), (TEXT_HTML), P_SYSTEM);
    });

//----------------------------------------------------------------------------

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40269d2f depc=0x00000000

LoadStoreError: Processor internal physical address or data error during load or store
  excvaddr=0x40269d2f in system_get_sdk_version at ??:?

>>>stack>>>
* 
<<<stack<<<

0x40269d2f in system_get_sdk_version at ??:?
0x4021fa41 in AsyncAbstractResponse::AsyncAbstractResponse(std::function<String (String const&)>) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebResponses.cpp:405
0x40279136 in system_get_sdk_version at ??:?
0x40269d2f in system_get_sdk_version at ??:?
0x4021facd in AsyncProgmemResponse::AsyncProgmemResponse(int, char const*, unsigned char const*, unsigned int, std::function<String (String const&)>) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebResponses.cpp:826
0x40269d2f in system_get_sdk_version at ??:?
0x40100c34 in malloc at /Users/daemons/platformio_packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:912
0x40279136 in system_get_sdk_version at ??:?
0x40269d2f in system_get_sdk_version at ??:?
0x4021cd24 in AsyncWebServerRequest::beginResponse(int, char const*, unsigned char const*, unsigned int, std::function<String (String const&)>) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebRequest.cpp:699
0x40229cc8 in String::equals(String const&) const at /Users/daemons/platformio_packages/framework-arduinoespressif8266/cores/esp8266/WString.cpp:543
 (inlined by) String::equals(String const&) const at /Users/daemons/platformio_packages/framework-arduinoespressif8266/cores/esp8266/WString.cpp:542
0x40229589 in String::copy(char const*, unsigned int) at /Users/daemons/platformio_packages/framework-arduinoespressif8266/cores/esp8266/WString.cpp:274
0x4020920c in sendHTMLResponse(AsyncWebServerRequest*, unsigned char const*, unsigned int, char const*, html_pages_t) at ./web_handlers/WServer.h:61
0x4021bf9e in __find_if<__gnu_cxx::__normal_iterator<String*, std::vector<String> >, __gnu_cxx::__ops::_Iter_pred<AsyncWebServerRequest::_removeNotInterestingHeaders()::<lambda(const String&)> > > at /Users/daemons/platformio_packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/stl_algobase.h:1959
 (inlined by) __find_if<__gnu_cxx::__normal_iterator<String*, std::vector<String> >, __gnu_cxx::__ops::_Iter_pred<AsyncWebServerRequest::_removeNotInterestingHeaders()::<lambda(const String&)> > > at /Users/daemons/platformio_packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/stl_algobase.h:1974
 (inlined by) find_if<__gnu_cxx::__normal_iterator<String*, std::vector<String> >, AsyncWebServerRequest::_removeNotInterestingHeaders()::<lambda(const String&)> > at /Users/daemons/platformio_packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/stl_algo.h:3934
 (inlined by) none_of<__gnu_cxx::__normal_iterator<String*, std::vector<String> >, AsyncWebServerRequest::_removeNotInterestingHeaders()::<lambda(const String&)> > at /Users/daemons/platformio_packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/stl_algo.h:471
 (inlined by) any_of<__gnu_cxx::__normal_iterator<String*, std::vector<String> >, AsyncWebServerRequest::_removeNotInterestingHeaders()::<lambda(const String&)> > at /Users/daemons/platformio_packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/stl_algo.h:490
 (inlined by) AsyncWebServerRequest::_removeNotInterestingHeaders() at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebRequest.cpp:156
0x40220b04 in AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at /Users/daemons/platformio_packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:622
0x40220f38 in AsyncWebServer::_attachHandler(AsyncWebServerRequest*) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebServer.cpp:152
0x4021e468 in AsyncWebServerRequest::_parseLine() at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebRequest.cpp:587
0x4021e5d0 in AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/daemons/platformio_libdeps/ESP8266/ESPAsyncWebServer/src/WebRequest.cpp:94
0x40103138 in wDev_ProcessFiq at ??:?
0x4024568b in ip_chksum_pseudo at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/inet_chksum.c:395
0x4023b48c in esp2glue_alloc_for_recv at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:478
[... cut ...]
0x402167ec in BackupRestore::send_Upload() at ./modules/backup_restore.h:166
0x40216c14 in loop at src/main.cpp:135
0x40104291 in lmacIsIdle at ??:?
0x401053d9 in lmacTxFrame at ??:?
0x401021f4 in pp_post at ??:?
0x40104fbe in lmacMSDUAged at ??:?
0x401002c4 in ets_post at /Users/daemons/platformio_packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:238
0x4010490d in lmacRecycleMPDU at ??:?
0x401021f4 in pp_post at ??:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------`

lumapu, 2024-09-03:
I think found the problem.

We're using the following prototype:
```cpp
AsyncWebServerResponse* AsyncWebServerRequest::beginResponse(int code, const char* contentType, const uint8_t* content, size_t len, AwsTemplateProcessor callback);
```

You're passing a `const char*` from `PROGMEM` type casted to `const char *` as the second argument to `beginResponse`. For me it feels that `beginResponse` can't consume a PROGMEM-const-char-ptr. 

I tried that by implementing your macros step by step to my code. Once I started passing `contentType` from PROGREM the issue occured. 

<details>
<summary>Stack trace</summary>

```
0x4025880b in system_get_sdk_version at ??:?
0x4021112d in AsyncAbstractResponse::AsyncAbstractResponse(std::function<String (String const&)>) at ??:?  
0x4025880b in system_get_sdk_version at ??:?
0x4021118c in AsyncProgmemResponse::AsyncProgmemResponse(int, char const*, unsigned char const*, unsigned int, std::function<String (String const&)>) at ??:?
0x4025a4fa in system_get_sdk_version at ??:?
0x4021004f in AsyncWebServerRequest::beginResponse(int, char const*, unsigned char const*, unsigned int, std::function<String (String const&)>) at ??:?
0x40100df0 in malloc at ??:?
0x40205d9b in DBGPRINT(String) at ??:?
0x4025a4fa in system_get_sdk_version at ??:?
0x4021005f in AsyncWebServerRequest::beginResponse(int, char const*, unsigned char const*, unsigned int, std::function<String (String const&)>) at ??:?
0x40205d9b in DBGPRINT(String) at ??:?
0x40205df8 in DBGPRINTLN(String) at ??:?
0x4025880b in system_get_sdk_version at ??:?
0x4025a4fa in system_get_sdk_version at ??:?
0x4021c770 in String::operator=(String const&) at ??:?
0x4025880b in system_get_sdk_version at ??:?
0x4020cc9b in Web::providePage(AsyncWebServerRequest*, unsigned char const*, unsigned int, char const*) at ??:?
0x4021c609 in String::copy(char const*, unsigned int) at ??:?
0x4020e8ec in _ZZN21AsyncWebServerRequest28_removeNotInterestingHeadersEvENKUlRK6StringE_clES2_$constprop$0 at WebRequest.cpp:?
0x402120bc in AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at ??:?
0x402124dc in AsyncWebServer::_attachHandler(AsyncWebServerRequest*) at ??:?
0x40210317 in AsyncWebServerRequest::_parseLine() at ??:?
0x40210442 in AsyncWebServerRequest::_onData(void*, unsigned int) at ??:?
0x40103028 in rcReachRetryLimit at ??:?
0x401034e2 in wDev_ProcessFiq at ??:?
0x4021fbb8 in AsyncClient::_recv(std::shared_ptr<ACErrorTracker>&, tcp_pcb*, pbuf*, int) at ??:?
0x4021fc18 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, int) at ??:?
0x4022e4a5 in tcp_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c:501 (discriminator 1)
0x40106b7c in umm_assimilate_down at umm_malloc.cpp:?
0x40233539 in ip4_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1467
0x4022b0a1 in mem_malloc at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:210
0x40100a6e in umm_free_core at umm_malloc.cpp:?
0x4024fe7c in ppRecycleRxPkt at ??:?
0x4022a765 in ethernet_input_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:188
0x4022a578 in git2glue_err at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:118
 (inlined by) esp2glue_ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:494
0x4025302d in ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:365
0x4025303f in ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:373
0x4024faab in ppPeocessRxPktHdr at ??:?
0x40254317 in ets_snprintf at ??:?
0x40105659 in call_user_start_local at ??:?
0x4010565f in call_user_start_local at ??:?
0x4010000d in call_user_start at ??:?
0x401000ab in app_entry_redefinable at ??:?
0x4024464c in cont_ret at cont.S.o:?
0x402445fd in cont_continue at cont.S.o:?
0x40250bff in pp_attach at ??:?
0x40250c4e in pp_attach at ??:?
0x401023ff in ppCalFrameTimes at ??:?
0x40238dee in ieee80211_output_pbuf at ??:?
0x4024fcc4 in ppTxPkt at ??:?
0x40238f57 in ieee80211_output_pbuf at ??:?
0x40105b9b in wdt_feed at ??:?
0x40100df0 in malloc at ??:?
0x4022a161 in glue2esp_linkoutput at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:301
[... 51 lines cut ...]
0x4025622a in wifi_set_opmode_current at ??:?
0x4020d46c in ESP8266WiFiGenericClass::mode(WiFiMode) at ??:?
0x40100dbc in free at ??:?
0x4020e7d5 in WiFiUDP::stop() at ??:?
0x40206210 in Wifi8266::tickNetworkLoop() at ??:?
0x40100dbc in free at ??:?
0x4021d75a in yield at ??:?
0x40205a57 in Scheduler::loop() at ??:?
0x4020a7ad in WifiAp::loop() at ??:?
0x4020a806 in Sonoff::loop() at ??:?
```
</details>

Then digging further I found out that the last argument of your function 
```cpp
void sendStaticResponse(AsyncWebServerRequest *request, const uint8_t *htmlContent, size_t htmlSize, const char *contentType);
```
has to be
```cpp
void sendStaticResponse(AsyncWebServerRequest *request, const uint8_t *htmlContent, size_t htmlSize, const _FlashStringHelper* contentType);
```

now you have to pass the `text/css` with your other macro `PSTRP(TEXT_CSS)`. With that change the code also works as expected.

In the end I don't know if this webserver library can be improved or your code has to be changed. I'm curious how this will be solved 😊

lumapu, 2024-09-03:
side note: my code size shrinked a few bytes using the macros, thanks for sharing 👍

```
before changing, passing content type using F(...)
RAM:   [====      ]  40.4% (used 33104 bytes from 81920 bytes)
Flash: [====      ]  39.8% (used 415773 bytes from 1044464 bytes)

after change using macros FPSTRP and PSTRP
RAM:   [====      ]  40.4% (used 33072 bytes from 81920 bytes)
Flash: [====      ]  39.8% (used 415357 bytes from 1044464 bytes)
```

mathieucarbou, 2024-09-03:
The other option is to use `String`, like it is done at a few other places in this lib for esp8266:

`beginResponse(int code, const String& contentType, const uint8_t* content, size_t len`

and 

`auto response = request->beginResponse(200, String(contentType), htmlContent, htmlSize);`

I don't think the library is at fault here because `beginResponse_P` variants were reported to work, and those only have a signature with `const String& contentType`, so an implicit conversion was done when calling it. The implicit conversion cannot be done with `beginResponse` since `beginResponse` is overloaded to support String and char*.