mathieucarbou/ESPAsyncWebServer #108: [BUG]

lumapu, 2024-09-29:
**Description**

Newest commit does not build with ArduinoJson

**Board**

esp32dev, esp32s3

**Ethernet adapter used ?**

no matter, compile time error

**Stack trace**

```
AsyncMessagePack.cpp
    'ArduinoJson::V6215PB2::JsonDocument::~JsonDocument()' is protected within this context
    "ArduinoJson::V6215PB2::JsonDocument::JsonDocument()" (declared at line 280 of "C:\lpusch\github\ahoy\src\.pio\libdeps\opendtufusion\ArduinoJson\src\ArduinoJson\Document\JsonDocument.hpp") is inaccessible
    no instance of overloaded function "ArduinoJson::V6215PB2::JsonDocument::add" matches the argument list
    object type is: ArduinoJson::V6215PB2::JsonDocument
    no matching function for call to 'ArduinoJson::V6215PB2::JsonDocument::add<ArduinoJson::V6215PB2::JsonArray>()'
    no instance of overloaded function "ArduinoJson::V6215PB2::JsonDocument::add" matches the argument list
    no matching function for call to 'ArduinoJson::V6215PB2::JsonDocument::add<ArduinoJson::V6215PB2::JsonObject>()'
    'ArduinoJson::V6215PB2::JsonDocument::~JsonDocument()' is protected within this context
    "ArduinoJson::V6215PB2::JsonDocument::JsonDocument()" (declared at line 280 of "C:\lpusch\github\ahoy\src\.pio\libdeps\opendtufusion\ArduinoJson\src\ArduinoJson\Document\JsonDocument.hpp") is inaccessible
AsyncMessagePack.h
    deleted function 'virtual AsyncMessagePackResponse::~AsyncMessagePackResponse()' overriding non-deleted function
```

**Additional notes**

Check my action which failed: https://github.com/lumapu/ahoy/actions/runs/11095964116/job/30825247520

lumapu, 2024-09-29:
last known sha which worked for me was `eacf0eb`.

mathieucarbou, 2024-09-29:
Hello @lumapu ,

Never point directly to `main`: use releases instead.

`https://github.com/mathieucarbou/ESPAsyncWebServer#v3.3.2`

`main` is used to accumulate features and fixes until stability / tests are done. This is a bleeding edge branch where there is no guarantee at all that everything works fine.

Which release you are using ? `eacf0eb` does not match any released version.

mathieucarbou, 2024-09-29:
I also see you are using ArduinoJson @ 6.21.5

mathieucarbou, 2024-09-29:
OK I understand what's wrong... This is because msg pack support was added but only for Arduino 7.
Will fix now.

closed 2024-09-29 by mathieucarbou

lumapu, 2024-09-30:
The point to `main` was by accident, I didn't know that the definition I had was not pointing to the release. 
Thank you for the quick response and fix, I hope you got enough rest last night 😉.

I use the ArduinoJson v6 internationally. I read the following note: `ArduinoJson 7 is significantly bigger`: https://arduinojson.org/v7/how-to/upgrade-from-v6/

mathieucarbou, 2024-09-30:
Well, I am glad you did that because you found a genuine bug ;-) Thanks!

commit b0541501 (2024-09-29): Merge pull request #109 from mathieucarbou/fix-108

Fix compilation issue with ArduinoJson 5 and 6 (closes #108)
commit e75dd5b0 (2024-09-29): Fix compilation issue with ArduinoJson 5 and 6 (closes #108)