mathieucarbou/ESPAsyncWebServer #111: [Q] checkBasicAuthentication calculates the base64 encoded hash for username:password everytime

stefan123t, 2024-09-30:
Wouldn't it be better to calculate the base64 encoded hash for `username:password` only once, when you modify it using setUsername and setPassword ?

I have seen in https://github.com/mathieucarbou/ESPAsyncWebServer/blob/main/src/WebAuthentication.cpp#L34-67 it is calculated for every WebRequest to the same AuthenticationMiddleware again.

mathieucarbou, 2024-09-30:
Hello @stefan123t : I've read the upstream discussion in OpenDTU project. I love this project:-) I have one setup myself ;-)

So for your question, it works as before, no change for that.

The issue with your proposal is that the authentication method (basic or digest) is decided by the request and not known in advance. If you test with curl for example, you can access a protected endpoint either with a basic or digest authentication headers 

The authentication method that is passed as parameter of the middleware is, like before, the authentication method that the server will use in its reply in case the authentication fails, in order to hint the browser (or curl).

I agree with your that considering now that the introduction of a middleware allows for more dynamic setup and also memory saving, maybe an option could be to either precomputed both hash, or just the one in relation to authentication method.

In any case I will have a look this evening if this can be quickly done.

stefan123t, 2024-09-30:
Salut @mathieucarbou thanks for your interest and reply.

For Basic Auth it can be stored in the AuthenticationMiddleware, whereas for Digest Auth it may depend a bit on how much of the Digest Auth details have been implemented in the ESPAsyncWebServer. I am not that familiar with Digest Auth, expecially when it comes to the advertising of the `nonce` / `cnonce` using a Server response.

I assume that the Server would have to make sure to announce a new `cnonce` at least for every `401 Unauthenticated` in order to prevent CORS and other Session high-jacking methods it should protect against. There is a quite detailled write-up on Wikipedia for more details: https://en.wikipedia.org/wiki/Digest_access_authentication

You can see that it makes use of `_hash1` and `_hash2`, which could be stored in the AuthenticationManager and only the second `_hash2` may need to be recomputed per request / handler. Just like an Apache WebServer also stores `_hash1` as part of the `.htdigest` file in the `<username>:<realm>:<_hash1>` format, here `_hash1` is the `md5sum( "username:realm:password")`. And `_hash2` is the `md5sum("<method>:</url/path/resource>")`. 
Only resulting the `response` is probably calculated new from request to request, i.e. `md5sum("<_hash1>:<nonce>:<nc>:<cnonce>:<qop>:<_hash2>")`, mostly because `nc`, `cnonce` and `nonce` may change over time, with `nc` being the most frequent IMHO, `cnonce` per Client and `nonce` less frequent.

commit 6a19dd06 (2024-09-30): Complete rework of AuthenticationMiddleware...
- to align methods and enum with PsychicHttp and Arduino WebServer
- to support hash
- to pre-compute base64 / digest hash to speed up requests
Closes #111
commit 414b4761 (2024-10-01): Complete rework of AuthenticationMiddleware...
- to align methods and enum with PsychicHttp and Arduino WebServer
- to support hash
- to pre-compute base64 / digest hash to speed up requests
Closes #111
commit 23fbdf84 (2024-10-01): Complete rework of AuthenticationMiddleware...
- to align methods and enum with PsychicHttp and Arduino WebServer
- to support hash
- to pre-compute base64 / digest hash to speed up requests
Closes #111
commit 3e416acc (2024-10-01): Complete rework of AuthenticationMiddleware...
- to align methods and enum with PsychicHttp and Arduino WebServer
- to support hash
- to pre-compute base64 / digest hash to speed up requests
Closes #111
closed 2024-10-01 by mathieucarbou