mathieucarbou/ESPAsyncWebServer #115: [Q] Working with chunked responses

minou65, 2024-10-02:
Hi all
The IotWebConf Class use Arduino's WebServer as default web server. But it has the possibility to create a new WebServerWrapper and using the ESPAsyncWebserver.

In the IotWebConf::handleConfig I can see that the html page is createt this way

```c++
    // Send chunked output instead of one String, to avoid
    // filling memory if using many parameters.
    webRequestWrapper->sendHeader( "Cache-Control", "no-cache, no-store, must-revalidate");
    webRequestWrapper->sendHeader("Pragma", "no-cache");
    webRequestWrapper->sendHeader("Expires", "-1");
    webRequestWrapper->setContentLength(CONTENT_LENGTH_UNKNOWN);
    webRequestWrapper->send(200, "text/html; charset=UTF-8", "");
    webRequestWrapper->sendContent(content);
    webRequestWrapper->sendContent(F(""));
    webRequestWrapper->stop();

```

I have created a class with the following methodes

```c++
	void sendHeader(const String& name, const String& value, bool first = false) override {
		_headers.push_back(new AsyncWebHeader(name.c_str(), value.c_str()));
		this->_first = first;

	};

	
         void send(int code, const char* content_type = nullptr, const String& content = String("")) override {
		AsyncWebServerResponse* response = this->_request->beginResponse(code, content_type, content.c_str());
		for (const auto& header : _headers) {
			response->addHeader(header->name().c_str(), header->value().c_str());
		};

		if (!isHeaderInList("Content-Length")) {
			response->setContentLength(this->_contentLength);
		}

		this->_request->send(response);

	};

	void sendContent(const String& content) override {
		if (content.length() == 0) {
			ESP_LOGW("AsyncWebRequestWrapper", "Warning: content is empty");
			return;
		}

		String convertedContent = content;
		convertedContent.replace("\r\n", "\n");

		try {
			std::vector<char> convertedContent(content.begin(), content.end());
			_content.insert(_content.end(), convertedContent.begin(), convertedContent.end());
		}
		catch (const std::exception& e) {
			ESP_LOGE("AsyncWebRequestWrapper", "Error: exception caught while adding content: %s", e.what());
			return;
		}
		catch (...) {
			ESP_LOGE("AsyncWebRequestWrapper", "Error: unknown exception caught while adding content");
			return;
		}
	}

	void stop() override {
		if (_content.empty()) {
			return;
		}

		try {
			auto cache_ = std::make_shared<std::vector<char>>(_content);

			AsyncWebServerResponse* response = this->_request->beginChunkedResponse("text/html", [cache_](uint8_t* buffer, size_t maxLen, size_t index) -> size_t {
				size_t len_ = std::min(cache_->size() - index, maxLen);
				if (len_ > 0 && index < cache_->size()) {
					memcpy(buffer, cache_->data() + index, len_);
					return len_;
				}
				else {
					return 0;
				}
				});

			response->addHeader("Server", "ESP Async Web Server");
			this->_request->send(response);
		}
		catch (const std::bad_alloc& e) {
			ESP_LOGE("AsyncWebRequestWrapper", "Error: bad_alloc exception caught");
		}
		catch (const std::exception& e) {
			ESP_LOGE("AsyncWebRequestWrapper", "Error: exception caught: %s", e.what());
		}
		catch (...) {
			ESP_LOGE("AsyncWebRequestWrapper", "Error: unknown exception caught");
		}
	}
```

It seems to work well for small HTML pages, but now I have a project with an HTML page around 50kB. And there, most of the time, I receive the following error:

`[ 31491][E][iotwebconfasyncclass.h:167] stop(): [AsyncWebRequestWrapper] Error: bad_alloc exception caught`

Freeheap shows me around 90kB free memory just before it crashes.

Is my concept correct? Why do I receive this error? What should I do better?

Regards

minou65, 2024-10-02:
On other question
Can I call
`this->_request->beginChunkedResponse`
as many time as needed and the send at the end
`this->_request->send(response)`

In this case a can send the chunks in the sendcontent method and in the stop methode I only send the closing command

mathieucarbou, 2024-10-02:
Hello,

The usage seems correct, but you have too much boilerplate code around to be capable of pinpointing the source of any issue right now.

Please work on a minimal reproductible use case only involving AsyncTCP lib and ESPAsyncWS directly, serving your big file through chunk encoding. And if you still have an issue, we'll look.

Also, please note that for file serving, if the file sits in FS, there is an existing API.

mathieucarbou, 2024-10-02:
Note: I will verify on my side with a little example if this is not broken....

mathieucarbou, 2024-10-02:
I added a chunk example, which works fine by the way, as we can see in the test below, 6 chunks are used to send 16k of data.

https://github.com/mathieucarbou/ESPAsyncWebServer/blob/main/examples/SimpleServer/SimpleServer.ino#L310-L325

```bash
❯  curl -N -v -X GET -H "origin: http://192.168.4.1" http://192.168.4.1/chunk
Note: Unnecessary use of -X or --request, GET is already inferred.
*   Trying 192.168.4.1:80...
* Connected to 192.168.4.1 () port 80
> GET /chunk HTTP/1.1
> Host: 192.168.4.1
> User-Agent: curl/8.10.0
> Accept: */*
> origin: http://192.168.4.1
> 
* Request completely sent off
< HTTP/1.1 200 OK
< Access-Control-Allow-Origin: http://192.168.4.1
< Access-Control-Allow-Methods: POST, GET, OPTIONS, DELETE
< Access-Control-Allow-Headers: X-Custom-Header
< Access-Control-Allow-Credentials: false
< Access-Control-Max-Age: 600
< Connection: close
< Accept-Ranges: none
< Transfer-Encoding: chunked
< Content-Type: text/html
< 
111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111 [... cut ...] 66666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666* shutting down connection #0
```

closed 2024-10-02 by mathieucarbou