mathieucarbou/ESPAsyncWebServer #126: [BUG] compilation error on ESP32C3 (use of deleted function)

mistic100, 2024-10-06:
**Description**

When compiling fot the Seeeduino Xiao ESP32C3 board, the following error happens:

```
c:\users\damien\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\ext\new_allocator.h:136:4: error: use of deleted function 'std::unique_ptr<_Tp, _Dp>::unique_ptr(const std::unique_ptr<_Tp, _Dp>&) [with _Tp = AsyncWebHandler; _Dp = std::default_delete<AsyncWebHandler>]'
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\users\damien\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\memory:80,
                 from C:/Users/Damien/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FS.h:24,
                 from C:/Users/Damien/.platformio/packages/framework-arduinoespressif32/libraries/SPIFFS/src/SPIFFS.h:17,
                 from src/Config.hpp:4,
                 from src/main.cpp:4:
c:\users\damien\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\bits\unique_ptr.h:397:7: note: declared here
       unique_ptr(const unique_ptr&) = delete;
       ^~~~~~~~~~
```

I don't really understand what is going on here, the stack trace does not originate from ESPAsyncWebServer be clearly references `AsyncWebHandler`.

If I remove all SPIFFS includes and usage, the same error happens at a different location:

```
c:\users\damien\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\ext\new_allocator.h:136:4: error: use of deleted function 'std::unique_ptr<_Tp, _Dp>::unique_ptr(const std::unique_ptr<_Tp, _Dp>&) [with _Tp = AsyncWebHandler; _Dp = std::default_delete<AsyncWebHandler>]'
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\users\damien\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\memory:80,
                 from C:/Users/Damien/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.h:26,
                 from C:/Users/Damien/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:37,
                 from src/WifiServer.hpp:3,
                 from src/main.cpp:5:
c:\users\damien\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\bits\unique_ptr.h:397:7: note: declared here
       unique_ptr(const unique_ptr&) = delete;
       ^~~~~~~~~~
```

**Link:** https://github.com/mistic100/esp-server-test

This is a simple server with the following features :
- AP mode
- serve HTML file from SPIFFS
- basic POST request
- JSON POST request

**Board**:  ESP32C3 

**Additional notes**

The error is not present when using `esphome/ESPAsyncWebServer-esphome` but this fork is less maintained and stuck to JsonArduino 6.

mathieucarbou, 2024-10-06:
I will have a look. Thanks!

mathieucarbou, 2024-10-06:
First you need to fix that:

```bash
> pio run
```

```
In file included from src/main.cpp:4:
src/Config.hpp:4:10: fatal error: SPIFFS.h: No such file or directory
```

SPDIFF has been deprecated and should not be used anymore/

mistic100, 2024-10-06:
Thanks for the patch, unfortunately I still get the same compilation error here.

Note that I don't get the error you have regarding missing SPIFFS lib, perhaps it as to do how packages are installed on Windows... don't know.

mistic100, 2024-10-06:
I was able to compile and run everything by using https://github.com/pioarduino/platform-espressif32 instead of the de default Espressif32 platform.

mathieucarbou, 2024-10-06:
Try to setup a GitHub workflow to checkout and buid your project on a git push.
It will be a way to assert if it builds correctly in a newly installed environment with the lastest tools.
But looking at your stack trace you seem to point to an older tooling so I would first check your tommong version.

mathieucarbou, 2024-10-06:
> No more luck on Linux.
> 
> However I was able to compile and run everything by using https://github.com/pioarduino/platform-espressif32 instead of the default Espressif32 platform.

Ok! This is what I thought. This is linked to the tooling version used under the hood.

There are some cache in ~/. platformio folder.... Maybe something is wrong or does not update ?

I am using pioarduino also a lot so when I did a Pio run on your project, maybe PlatformIO found some more recent tooling in my cache.

You should use pioarduino too because PlatformIO does not support Arduino anymore and the only way to access recent Arduino cord tooling (3.x) is by using pioarduino.