mathieucarbou/ESPAsyncWebServer #128: [BUG] ArduinoJson initialization fails if using Arduino IDE instead of PlatformIO

nomakewan, 2024-10-07:
**Description**

Thanks to release 3.3.12 finally being picked up by the Arduino Library Manager I was able to update from 3.0.6. After then updating AsyncTCP to 3.2.8 as well, I found that my code would error out with the following compile error:

`Webhandler.h:190:1: error: 'JsonDocument' does not name a type`

Checking the diffs between 3.3.12 and 3.0.6, the major difference was in `AsyncJson.h`, where this code was found:

```cpp
#if __has_include("ArduinoJson.h")
  #include <ArduinoJson.h>
  #if ARDUINOJSON_VERSION_MAJOR >= 5
    #define ASYNC_JSON_SUPPORT 1
  #else
    #define ASYNC_JSON_SUPPORT 0
  #endif // ARDUINOJSON_VERSION_MAJOR >= 5
#endif   // __has_include("ArduinoJson.h")
```

My guess is that when compiling with the Arduino IDE, that first if statement is failing, which is causing everything within to be skipped (including initializing the ArduinoJson library). When I change the above section to the following, everything compiles fine:

```cpp
#include <ArduinoJson.h>
#define ASYNC_JSON_SUPPORT 1
```

I am using ArduinoJson 7.2.0, in case that helps.

**Link:** https://github.com/gpstar81/GPStar-proton-pack/blob/v5.3.4/develop/source/AttenuatorESP32/include/Webhandler.h#L190

**Board**: esp32dev

**Ethernet adapter used ?** No

**Stack trace**

Cannot provide stack trace as compilation fails.

**Additional notes**

N/A

mathieucarbou, 2024-10-07:
I don't really understand: your project is a platformio project but you are talking about the arduino library registry, which platformio is not using... Pio is looking at pio registry (https://registry.platformio.org).

you are using ArduinoJson, so `#if __has_include("ArduinoJson.h")` should be true.

Try to set your includes ordering right:

```c++
#include <Arduino.h>
#include <ESPAsyncWebServer.h>
```

you do not need to set ASYNC_JSON_SUPPORT to 1.

All this complexity is to handle the optional Json code  so that it activates wen the library is inside the compile path.

mathieucarbou, 2024-10-07:
also, maybe this could help you:

```c++
lib_compat_mode = strict
lib_ldf_mode = chain
```

Though, the project examples compile fine without it...

nomakewan, 2024-10-07:
By default yes, it's a PlatformIO project.

But I'm locally compiling with the Arduino IDE (by moving the contents of the `includes` folder into `src` and renaming `main.cpp` to `src.ino`.

I'll give that includes order change a shot right now.

mathieucarbou, 2024-10-07:
I really think this is an include issue because if I load the project sample:

https://github.com/mathieucarbou/ESPAsyncWebServer/blob/main/examples/SimpleServer/SimpleServer.ino

In Arduino IDEA with the latest libs, it works.

Even if I remove all the `#if ASYNC_JSON_SUPPORT == 1` which are only there to be able to build the sample in CI with and without json support. In a normal project, nobody dhould use these `#if ASYNC_JSON_SUPPORT == 1`

nomakewan, 2024-10-07:
Changing the order of includes did not work. I then changed the `AsyncJson.h` startup to be this:

```cpp
#if __has_include("ArduinoJson.h")
  #include <ArduinoJson.h>
  #if ARDUINOJSON_VERSION_MAJOR >= 5
    #define ASYNC_JSON_SUPPORT 1
  #else
    #define ASYNC_JSON_SUPPORT 0
  #endif // ARDUINOJSON_VERSION_MAJOR >= 5
#else
  #ifdef __has_include
    #error "__has_include defined, but library not found!"
  #else
    #error "__has_include not defined!"
  #endif
#endif   // __has_include("ArduinoJson.h")
```

And the compiler produced the output `47 |     #error "__has_include defined, but library not found!"`.

Could this be related to the issue being discussed here, regarding how the Arduino IDE handles directory structures for libraries?

https://forum.pjrc.com/index.php?threads/use-if-__has_include-for-something-like-user_defs-h-in-arduino-sketch.60747

They mention how PlatformIO seems to work perfectly fine but the Arduino IDE builds fail due to a path issue.

mathieucarbou, 2024-10-07:
I frankly don't know.... It works on my computer with the project sample...

Maybe an issue with your Arduino lib folder ?

Arduino is doing shady things behind to mount the project compilation flags, when PIO allows you to have a complete control (and reliable one) over that.

nomakewan, 2024-10-07:
Tested the project sample and had the exact same compilation error as I was having with our code.

It occurs to me though, I'm using Arduino IDE 1.8.19. I wonder if that changes something?

Again, if I remove the `#if __has_include("ArduinoJson.h")` and just do `#include <ArduinoJson.h>` instead, it compiles just fine.

mathieucarbou, 2024-10-07:
> It occurs to me though, I'm using Arduino IDE 1.8.19. I wonder if that changes something?

Ah definitely! You should always upgrade to the latest Arduino version because the tooling behind updates also. Maybe the default compile flags are not the same...

Arduino CI will also test against the latest arduino version...

mathieucarbou, 2024-10-13:
@nomakewan : can we close this issue ? you were able to solve your IDE problem ?

nomakewan, 2024-10-14:
I am not able to resolve the IDE problem without switching operating systems. However, if the intent of ESPAsyncWebServer is to only serve either PlatformIO or the latest Arduino IDE, then my problem is irrelevant. As such, I will close this issue. Thank you.

closed 2024-10-14 by nomakewan