mathieucarbou/ESPAsyncWebServer #142: [BUG] WebSocket connection not upgraded when URLs are same

ayushsharma82, 2024-10-22:
@mathieucarbou , Recently I was integrating support for multiple instances of WebSerial. The problem I'm facing is that AsyncWebSocket fails to work if we also have an HTTP route at same URL. I guess, AsyncWebServer is not honouring the `Upgrade: websocket` header when both are on same URL paths.

The following fails:
HTTP Route URL: `/webserial`
WS URL: `/webserial` or even `/webserial/pipe`

The following works:
HTTP Route URL: `/webserial`
WS URL: `/wsl`

ayushsharma82, 2024-10-22:
Okay, some good observations and not that big of a bug.

### Case 1 (success):
It works fine if we add the AsyncWebSocket handler (on same URL path) before HTTP handler.

```c++
_server->addHandler("/webserial");

_server->on("/webserial", HTTP_GET, [&](AsyncWebServerRequest *request){
// ...
});
```

### Case 2 (fails):
It fails to work if the websocket handler is added after HTTP handler.

```c++
_server->on("/webserial", HTTP_GET, [&](AsyncWebServerRequest *request){
// ...
});

_server->addHandler("/webserial");
```

In the second case, HTTP handler will grab & serve the request before reaching the WS handler which is not the right move . This is a problem in the `canHandle` function logic, it should _NOT_ grab a `GET` request meant for asyncwebsocket based on `Upgrade` header.

mathieucarbou, 2024-10-22:
@ayushsharma82 : I'll see what I can do. This behaviour is there since years if I am not mistaken and did not change since the original project so fixing that could break some existing app relying on this ordering right ?

I also noticed yes that there is an ordering to match when defining the http handlers, but did not notice yet that with ws.

ayushsharma82, 2024-10-22:
True. I noticed this with HTTP routes too earlier, this is indeed a bug which had to be fixed at some point. This WS thing makes it worse because that request should never be caught by an HTTP handler.

I don't think so fixing it will break any existing app... Nobody would have ordered the routes in such a way that would block their own functionality.

mathieucarbou, 2024-10-22:
I am looking at the code right now and when a request is parsed, if the `Upgrade` header is there and set to `websocket`, then the flag `RCT_WS` is set.

The canHandle of the ws handler is written as follow:

```c++
bool AsyncWebSocket::canHandle(AsyncWebServerRequest* request) {
  if (!_enabled)
    return false;

  if (request->method() != HTTP_GET || !request->url().equals(_url) || !request->isExpectedRequestedConnType(RCT_WS))
    return false;

  return true;
}
```

The ones for default http handlers starts with:

```c++
bool AsyncCallbackWebHandler::canHandle(AsyncWebServerRequest* request) {
  if (!_onRequest)
    return false;

  if (!(_method & request->method()))
    return false;
```

So your suggestion is that we should also return false if `request->isExpectedRequestedConnType(RCT_WS)` is true ?

```c++
bool AsyncCallbackWebHandler::canHandle(AsyncWebServerRequest* request) {
  if (!_onRequest)
    return false;

  if (!(_method & request->method()))
    return false;

  if (request->isExpectedRequestedConnType(RCT_WS))
    return false;
```

Line 270 of WebHandlers.cpp

This patch would make sense to me... Could you test it to confirm ?

ayushsharma82, 2024-10-22:
@mathieucarbou  Tested and works fine 👍🏻

mathieucarbou, 2024-10-22:
Ok! I will add the patch to all other implementations and release tomorrow

commit 724f6c30 (2024-10-23): Fix #142: canHandle implementations should check for the request type (http, websocket, sse)
closed 2024-10-23 by mathieucarbou