mathieucarbou/ESPAsyncWebServer #144: param-rewrite-with-matching does not longer compile with v3.3.19

mathieucarbou, 2024-10-24:
### Discussed in https://github.com/mathieucarbou/ESPAsyncWebServer/discussions/143

<div type='discussions-op-text'>

<sup>Originally posted by **tueddy** October 24, 2024</sup>
I have param-rewrite code exactly like in documention:
https://github.com/mathieucarbou/ESPAsyncWebServer?tab=readme-ov-file#param-rewrite-with-matching

With 3.3.19 the code no longer compiles:

`error: 'bool OneParamRewrite::match(const AsyncWebServerRequest*)' marked 'override', but does not override
  218 |     bool match(const AsyncWebServerRequest* request) override {
      |          ^~~~~
`

Can you show the new match() function and update the docs?
Thanks!</div>

mathieucarbou, 2024-10-24:
Fixed in v3.3.20

closed 2024-10-24 by mathieucarbou

tueddy, 2024-10-24:
Works as expected now, many thanks for the fast fix!

mathieucarbou, 2024-10-24:
@tueddy : i've reviewed the original doc and I `OneParamRewrite` is flawed be careful. It does not handle the case when a hash or query param is added to the URL, or else.

`server.addRewrite( new OneParamRewrite("/radio/{frequence}", "/radio?f={frequence}") );`

Example, if browser sends: `/radio/{frequence}?v=1` then the result will be `/radio?f={frequence}?v=1`