mathieucarbou/ESPAsyncWebServer #151: [BUG] setAuthentication on serveStatic not working from 3.3.x

rostwolke, 2024-11-04:
**Description**

First, thanks for the good work! After switching from the me-no-dev version of the repo I noticed that my web ui does not ask me for authentication anymore. The code to reproduce this is simply:
```
server->serveStatic("/", SPIFFS, "/www/").setDefaultFile("index.html").setAuthentication("test", "test");
```

I checked when the problem occurred for the first time. I switched versions back in arduino and the 3.0.6 is the first one that seems to work again.

**Link:** See example above.

**Board**: ESP32-C3 Super Mini

**Ethernet adapter used ?** No

**Stack trace**

Should not be necessary since the ESP does not crash.

**Additional notes**

I noticed that this version of the async webserver supports middleware. I am sure I can fix my problem using middleware. I still wanted to inform you since setAuthentication is still listed in the docs.

mathieucarbou, 2024-11-04:
Thanks for the bug report! Indeed, I saw a bug which was introduced when adding backward compatibility for middleware support. I will fix now.

mathieucarbou, 2024-11-04:
A workaround is to use middleware

closed 2024-11-04 by mathieucarbou

rostwolke, 2024-11-05:
@mathieucarbou Works like a charm. Thanks again and keep up the good work :-)

commit 233d0ee1 (2024-11-04): Merge pull request #152 from mathieucarbou/issue-151

Fix #151: setAuthentication() was not authenticating: the middleware …
commit 7cf5de6e (2024-11-04): Fix #151: setAuthentication() was not authenticating: the middleware was setup with `AUTH_NONE` (default) instead of `AUTH_DIGEST` (to keep backward compatibility)