mathieucarbou/ESPAsyncWebServer #153: [Q] [String] 'username=" ... a13ed6f5",': Reallocating large String(239 -> 240 bytes)

DaeMonSxy, 2024-11-08:
It seems after using: 
```
-DDEBUG_ESP_PORT=Serial         ; Sets the Serial port for debug output
-DDEBUG_ESP_WIFI               ; Enables detailed WiFi debugging
-DCORE_DEBUG_LEVEL=5 ; Verbose
```

i see following msgs:
```
[String] 'username=" ... a13ed6f5",': Reallocating large String(239 -> 240 bytes)
[String] 'username=" ... a77664a5",': Reallocating large String(239 -> 240 bytes)
```
in case i would refresh page, 
```
[String] 'username=" ... 1149d95a",': Reallocating large String(239 -> 240 bytes)
2736
ESP.getFreeHeap <= 3000 restart
2736
ESP.getFreeHeap <= 3000 restart
2736
ESP.getFreeHeap <= 3000 restart
2736
ESP.getFreeHeap <= 3000 restart
2608
ESP.getFreeHeap <= 3000 restart
2608
ESP.getFreeHeap <= 3000 restart
2608
ESP.getFreeHeap <= 3000 restart
2480
ESP.getFreeHeap <= 3000 restart
2512
ESP.getFreeHeap <= 3000 restart
2992
ESP.getFreeHeap <= 3000 restart
2992
ESP.getFreeHeap <= 3000 restart
2992
ESP.getFreeHeap <= 3000 restart
2992
ESP.getFreeHeap <= 3000 restart
2992
ESP.getFreeHeap <= 3000 restart
2992
ESP.getFreeHeap <= 3000 restart
[String] 'username=" ... 7b8dd4d0",': Reallocating large String(239 -> 240 bytes)
```
so, something is complatelly chuwing the heap,then reset.

tried to dig in deeper, found:
[https://github.com/esp8266/Arduino/issues/8872](https://github.com/esp8266/Arduino/issues/8872)

so, anything we can do within async-webserver?

mathieucarbou, 2024-11-08:
Strings in ESPAsyncWebServer are not anymore on flash for ESP82666 but in RAM since the introduction of the namespace class. So heap usage is increased.

**Here is the warning for ESP8266:** 

With @vortigont we will not put any efforts in maintaining this lib for ESP8266. We focus on ESP32.
**Even if we try to keep it backward compatible, there won't be any improvements or fix made.**
ESP8266 goes EOL in a few years and there is no development anymore. All ESP8266 can be replaced by equivalent ESP32 boards.

People who really want to stick with ESP8266 will have to stick with an older version of the library, or the original one.

**Version 4.x of this ESPAsyncWebServer fork will drop support for ESP8266.**

DaeMonSxy, 2024-11-08:
ok,
got it,
I do have a few (above 10 ) relays around the house based on esp8266...so would prefer to keep them working until functioning.

Any addvice to get rid of that issue for now?

vortigont, 2024-11-08:
> Version 4.x of this ESPAsyncWebServer fork will drop support for ESP8266

can't wait for this day to come :)

> Any addvice to get rid of that issue for now?

it can't be avoided, happens when concatenating long strings in many places in WS code. The only way to prevent this is to reserve large string buffer in advance, but it will put more pressure to memory anyway.

But we will keep this in mind in future code optimizations, thanks!

mathieucarbou, 2024-11-20:
> [String] 'username=" ... 7b8dd4d0",': Reallocating large String(239 -> 240 bytes)

@DaeMonSxy : looking at that again, pretty sure such concatenation does not come from ESPAsyncWS code: there is no string concatenation happening with username=...

I will close this issue.

closed 2024-11-20 by mathieucarbou

DaeMonSxy, 2024-11-21:
@mathieucarbou  ok, no problem.
However, would be good to know, where the messages comes from:

```
[String] 'username=" ... eb41d51c7"': Reallocating large String(238 -> 240 bytes)
[String] 'HTTP/1.1 2 ... es: none
```

This is the message at the time of loading a webpage, so this should come from webserver or asynctcp -I think.
Didnt find refference though.

I believe the username= msg linked to this lines:
```
 if (dev_auth && !request->authenticate(dev_user, dev_pass)) {
        request->requestAuthentication();
        return;
    }
```