mathieucarbou/ESPAsyncWebServer #161: Example not working on ESP32-C3

NOV62, 2024-11-18:
Hello! The Captive Portal example does not work on the ESP32-C3 Supermini board. It compiles without errors, loads, an access point appears and that's it. Nothing happens when you try to log in to the address 192.168.4.1. I tried it on 3 modules. Arduino IDE 2.3.3. What could be the matter?

mathieucarbou, 2024-11-18:
I can reproduce on esp32dev... 
Looking... I have many apps using CP and I do not code it like this example which comes from the original fork.

mathieucarbou, 2024-11-18:
Fixed: https://github.com/mathieucarbou/ESPAsyncWebServer/commit/33611e8c0afd00f2ad4d115fe0ac65e3785851de

What happened is that the base method signature was changed to add `const`, but the overridden methods not having `override`, the compiler was unable to hint about the signature changes and canHandle was returning false (from base method).

NOV62, 2024-11-19:
Thanks for the tip. It became clear why some of my sketches don't work. I'm just learning.

closed 2024-11-19 by NOV62