mathieucarbou/ESPAsyncWebServer #162: [BUG] ws.closeAll() causes a crash

dewenni, 2024-11-20:
**Description**

It looks to me like there is a problem with the ws.closeAll() function. If you call the function while the websocket connection is still active, a crash occurs.

**Board**: esp32dev

**Stack trace**

```
Client #1 connected.
Closing all WebSocket clients...
All clients closed.

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d5753  PS      : 0x00060630  A0      : 0x800d5994  A1      : 0x3ffd22a0  
A2      : 0x3ffd32f8  A3      : 0x3ffd6336  A4      : 0xffffe40e  A5      : 0xba9cb054  
A6      : 0x3ffd3378  A7      : 0xfa9a138a  A8      : 0x50ae82e2  A9      : 0xba9cb054  
A10     : 0x0054b000  A11     : 0x0000007e  A12     : 0x00000004  A13     : 0x3ffd3360  
A14     : 0x3ffd62bd  A15     : 0x0000006f  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfa9a138a  LBEG    : 0x4008aad5  LEND    : 0x4008aaf7  LCOUNT  : 0x00000000  

Backtrace: 0x400d5750:0x3ffd22a0 0x400d5991:0x3ffd22e0 0x4015eb05:0x3ffd2300 0x4015eb75:0x3ffd2330 0x4015f516:0x3ffd2350 0x4008f762:0x3ffd2390
  #0  0x400d5750 in AsyncWebSocketClient::_onData(void*, unsigned int) at .pio/libdeps/esp32/ESPAsyncWebServer/src/AsyncWebSocket.cpp:529
  #1  0x400d5991 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(void*, AsyncClient*, void*, unsigned int)#5}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, void*&&, unsigned int&&) at .pio/libdeps/esp32/ESPAsyncWebServer/src/AsyncWebSocket.cpp:288
      (inlined by) __invoke_impl<void, AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::<lambda(void*, AsyncClient*, void*, size_t)>&, void*, AsyncClient*, void*, unsigned int> at /Users/toe/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/invoke.h:61
      (inlined by) __invoke_r<void, AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::<lambda(void*, AsyncClient*, void*, size_t)>&, void*, AsyncClient*, void*, unsigned int> at /Users/toe/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/invoke.h:111
      (inlined by) _M_invoke at /Users/toe/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/std_function.h:290
  #2  0x4015eb05 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at /Users/toe/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/std_function.h:591
      (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp32/AsyncTCP/src/AsyncTCP.cpp:1024
  #3  0x4015eb75 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp32/AsyncTCP/src/AsyncTCP.cpp:1400
  #4  0x4015f516 in _async_service_task(void*) at .pio/libdeps/esp32/AsyncTCP/src/AsyncTCP.cpp:190
      (inlined by) _async_service_task at .pio/libdeps/esp32/AsyncTCP/src/AsyncTCP.cpp:225
  #5  0x4008f762 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 044b2a97e6e6f015

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
```

**Additional notes**

I have created a small minimal example which is not a real use case, but with which it can be reliably simulated. I think closing an active connection should not lead to a crash in this case, even if it would not make sense.

```c++
AsyncWebServer server(80);
AsyncWebSocket ws("/ws");

void onWsEvent(AsyncWebSocket *server, AsyncWebSocketClient *client, AwsEventType type, void *arg, uint8_t *data, size_t len) {
  if (type == WS_EVT_CONNECT) {
    Serial.printf("Client #%u connected.\n", client->id());
  } else if (type == WS_EVT_DISCONNECT) {
    Serial.printf("Client #%u disconnected.\n", client->id());
  }
}

void setup() {
  Serial.begin(115200);

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  Serial.println("Connected to WiFi");

  ws.onEvent(onWsEvent);
  server.addHandler(&ws);

  server.on("/close_all_ws_clients", HTTP_POST, [](AsyncWebServerRequest *request) {
    Serial.println("Closing all WebSocket clients...");
    ws.closeAll();
    request->send(200, "application/json", "{\"status\":\"all clients closed\"}");
    Serial.println("All clients closed.");
  });

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    request->send(200, "text/html", R"rawliteral(
    <!DOCTYPE html>
        <html>
        <head></head>
        <script> 
            let ws = new WebSocket("ws://" + window.location.host + "/ws");
                    
            document.addEventListener("DOMContentLoaded", function () {
                ws.onopen = function () {
                    console.log("WebSocket connected");
                    };
                });
                        
            function closeAllWsClients() {
              fetch("/close_all_ws_clients", {
                  method : "POST",
              });
            };
        </script>
        <body>
            <button onclick = "closeAllWsClients()" style = "width: 200px"> ws close all</button><p></p>  
          </body>
        </html>
  )rawliteral");
  });

  server.begin();
}
```

mathieucarbou, 2024-11-20:
Thanks a lot for the code! I am trying now...

mathieucarbou, 2024-11-20:
> Serial.println("All clients closed.");

Pay attention that this is wrong: when send() is called, the request is NOT sent ;-) It is just a way to setup the status code, content type and body in the response object.

the data is sent over the network AFTER the middleware chain processing ;-)

I am not able to reproduce with Chrome: I am opening http://192.168.4.1/ and then go to: http://192.168.4.1/close_all_ws_clients

Note: I am using AP mode and `HTTP_GET | HTTP_POST` for `/close_all_ws_clients`: I added your example here: https://github.com/mathieucarbou/ESPAsyncWebServer/blob/main/examples/Issue162/Issue162.ino

mathieucarbou, 2024-11-20:
The stacktrace crashes at  `AsyncWebSocket.cpp:288`

`_client->onData([](void* r, AsyncClient* c, void* buf, size_t len) { (void)c; ((AsyncWebSocketClient*)(r))->_onData(buf, len); }, this);`

this is when the client is receiving some data. 
But I see no websocket data sent from the HTML content to the server ?

dewenni, 2024-11-20:
No I don't sent ws data.
You can open the root page to connect WebSocket and then press the button.
The button will redirect to "/close_all_ws_clients"

mathieucarbou, 2024-11-20:
> No I don't sent ws data. You can open the root page to connect WebSocket and then press the button. The button will redirect to "/close_all_ws_clients"

tried that also: no crash. So I tried something else.

dewenni, 2024-11-20:
hm, interesting. I have now also testet 100% your new example code :
https://github.com/mathieucarbou/ESPAsyncWebServer/blob/main/examples/Issue162/Issue162.ino

and it crashes on every try:  

[image]

mathieucarbou, 2024-11-20:
oh boy you are right! I can reproduce it with Safari but not Chrome! What the hell...

mathieucarbou, 2024-11-20:
Ok I have each time different kind of crash... It looks like a memory corruption. For example here is what I got. It is like Safari sends some data to the ws channel, or ESPAsyncWZ reads something...

```
Client #1 connected.
Closing all WebSocket clients...
Client #1 data: 
Client #1 data: �?
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: _T��7�
                      �T���
Client #1 data: �@p��?D
                       �?xV��0
Client #1 data:  IOelHvacHtS 007
5__ AKle?^bKO/6X.1F
5 @pHT%w, RkeH|ecT) >^rs�i���nh�+��+�� h�
                                         �k/h�?h����#ד�S�g�[��h
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: �?�
Client #1 data: 1.15
Client #1 data: 
Client #1 data: S�*|��*|���H�R*m<|��nz-6|��*|��
Client #1 data: p��?h
                     �?xV��1
Client #1 data: 
Client #1 data: ��\
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: �?������?��?
Client #1 data: 
Client #1 error.
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: ���E������
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
Client #1 data: 
lient #1 data: 
Client #1 data: 
Client #1 data: 
[... 281 lines cut ...]
[   795][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 101 - WIFI_READY
[   928][V][AP.cpp:108] _onApEvent(): AP Started
[   934][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 130 - AP_START
[   942][V][AP.cpp:88] _onApArduinoEvent(): Arduino AP Event: 130 - AP_START
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   334228 B ( 326.4 KB)
  Free Bytes        :   242936 B ( 237.2 KB)
  Allocated Bytes   :    81316 B (  79.4 KB)
  Minimum Free Bytes:   242660 B ( 237.0 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     3 : UART_RX[0]
============ After Setup End =============
[  1881][V][AP.cpp:125] _onApEvent(): AP Station Connected: MAC: be:15:c7:3d:11:da, AID: 1
[  1890][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 132 - AP_STACONNECTED
[  1900][V][AP.cpp:88] _onApArduinoEvent(): Arduino AP Event: 132 - AP_STACONNECTED

```

mathieucarbou, 2024-11-20:
My code had a mistake: I should not print `data` content because it contains binary data. I have pushed the fix, now I get:

```
[  4567][V][AP.cpp:125] _onApEvent(): AP Station Connected: MAC: be:15:c7:3d:11:da, AID: 1
[  4577][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 132 - AP_STACONNECTED
[  4586][V][AP.cpp:88] _onApArduinoEvent(): Arduino AP Event: 132 - AP_STACONNECTED
Client #1 connected.
Closing all WebSocket clients...
Client #1 len: 0
Client #1 error.
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 64
Client #1 len: 124
Client #1 len: 111
Client #1 len: 0
Client #1 len: 18
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 33
Client #1Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4009159f  PS      : 0x00060e33  A0      : 0x800910d7  A1      : 0x3ffb3280  
A2      : 0x3ffb0014  A3      : 0x3ffb20d0  A4      : 0xbaad5678  A5      : 0xbaad5678  
A6      : 0x3ffc41cc  A7      : 0x3ffb20c8  A8      : 0x00000007  A9      : 0xbe78de85  
A10     : 0x3ffb2140  A11     : 0x0000001a  A12     : 0x15c2cc99  A13     : 0x00000004  
A14     : 0x00000000  A15     : 0x0000cdcd  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x15c2cca5  LBEG    : 0x400898fc  LEND    : 0x40089912  LCOUNT  : 0xffffffff  

Backtrace: 0x4009159c:0x3ffb3280 0x400910d4:0x3ffb32a0 0x40091255:0x3ffb32c0 0x40083653:0x3ffb32e0 0x40092579:0x3ffb3300 0x400fede7:0x3ffb3320 0x400ee562:0x3ffb3340 0x400ef7e1:0x3ffb3360 0x400ef7fb:0x3ffb3380 0x400efcfc:0x3ffb33a0 0x400efd4f:0x3ffb33c0 0x400f2093:0x3ffb33e0 0x400f73ca:0x3ffb3420 0x400fc206:0x3ffb3450 0x400fe8dd:0x3ffb3470 0x4008cd6e:0x3ffb34a0
  #0  0x4009159c in remove_free_block at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/tlsf/tlsf.c:331 (discriminator 1)
      (inlined by) block_remove at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/tlsf/tlsf.c:380 (discriminator 1)
      (inlined by) block_merge_next at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/tlsf/tlsf.c:486 (discriminator 1)
      (inlined by) tlsf_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/tlsf/tlsf.c:1204 (discriminator 1)
  #1  0x400910d4 in multi_heap_free_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:235
  #2  0x40091255 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:286
  #3  0x40083653 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps_base.c:70
  #4  0x40092579 in free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c:39
  #5  0x400fede7 in mem_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/mem.c:236
  #6  0x400ee562 in pbuf_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c:780
      (inlined by) pbuf_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c:725
  #7  0x400ef7e1 in tcp_seg_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:1638
  #8  0x400ef7fb in tcp_segs_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:1623
  #9  0x400efcfc in tcp_pcb_purge at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:2204
      (inlined by) tcp_pcb_purge at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:2169
  #10 0x400efd4f in tcp_pcb_remove at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:2226
  #11 0x400f2093 in tcp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:459
  #12 0x400f73ca in ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:752
  #13 0x400fc206 in ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:186
  #14 0x400fe8dd in tcpip_thread_handle_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:174
      (inlined by) tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:148
  #15 0x4008cd6e in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 158ff685e

Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr 
Core  0 register dump:
PC      : 0x40084720  PS      : 0x00000016  A0      : 0x40080306  A1      : 0x3ffc5c90  
A2      : 0x3ffb315c  A3      : 0x00000000  A4      : 0x00100590  A5      : 0x3ffc5cc0  
A6      : 0x00000000  A7      : 0x3ffc5850  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00008000  A12     : 0x3ffb3100  A13     : 0x70000005  
A14     : 0xfffbfff0  A15     : 0x3ffc3f1c  SAR     : 0x00000008  EXCCAUSE: 0x00000001  
EXCVADDR: 0x15c2cca5  LBEG    : 0x400898fc  LEND    : 0x40089912  LCOUNT  : 0xffffffff  

Backtrace: 0x4008471d:0x3ffc5c90 0x40080303:0x3ffc5cc0 0x65a5a5a2:0xa5a5a5a5 |<-CORRUPTED
```

mathieucarbou, 2024-11-20:
@vortigont : FI, this one is a complicated one.

From the logs above:

> Client #1 error.

Something is off in the WS code when there is a disconnection and a mix of error / data (data probably sent by the browser)

vortigont, 2024-11-21:
interesting... I'll try to look into this also on a weekend. Do not have Safari at hand though :((( but will try with other clients

mathieucarbou, 2024-11-25:
Yes this seems to be only related to firefox and when the connexion is closed from the MCU.

It generates many different stack traces.

My assumption so far is that this is a concurrent issue because when the websocket connexion is opened, even if the use does not send some data, the browser will still send the control packets (ping/pong). We see them sometimes in the logs before the crash.

When closing the connexion from the MCU, the code who is closing might execute at the same time the data is being received or sent and that is where it crashes.

I think fixing this issue is just a matter of maybe adding a lock at the right place, but I am insure where at the moment.

vortigont, 2024-11-26:
yeah, probably some use-after-free pointer or so. Just do not get how Safari is affected here or how to get this on some other client

dewenni, 2024-11-29:
In this context, ws.cleanupClients() is also affected.

```c++
AsyncWebServer server(80);
AsyncWebSocket ws("/ws");

void onWsEvent(AsyncWebSocket *server, AsyncWebSocketClient *client,
               AwsEventType type, void *arg, uint8_t *data, size_t len) {
  if (type == WS_EVT_CONNECT) {
    Serial.printf("Client #%" PRIu32 " connected.\n", client->id());
  } else if (type == WS_EVT_DISCONNECT) {
    Serial.printf("Client #%" PRIu32 " disconnected.\n", client->id());
  } else if (type == WS_EVT_ERROR) {
    Serial.printf("Client #%" PRIu32 " error.\n", client->id());
  } else if (type == WS_EVT_DATA) {
    Serial.printf("Client #%" PRIu32 " len: %u\n", client->id(), len);
  }
}

void setup() {
  Serial.begin(115200);

  WiFi.mode(WIFI_AP);
  WiFi.softAP("esp-captive");

  ws.onEvent(onWsEvent);
  server.addHandler(&ws);

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    request->send(200, "text/html", R"rawliteral(
    <!DOCTYPE html>
        <html>
        <head></head>
        <script> 
            let ws = new WebSocket("ws://" + window.location.host + "/ws");
                    
            document.addEventListener("DOMContentLoaded", function () {
                ws.onopen = function () {
                    console.log("WebSocket connected");
                    };
                });
        </script>
        <body>
            <p>TEST-PAGE</p>  
          </body>
        </html>
  )rawliteral");
  });

  server.begin();
}

void loop() { 
  ws.cleanupClients(2); 
  }
```

[image]

dewenni, 2024-11-29:
I tried to prevent that more than MAX_WS_CLIENTS be opened with this code:

```c++
void onWsEvent(AsyncWebSocket *server, AsyncWebSocketClient *client,
               AwsEventType type, void *arg, uint8_t *data, size_t len) {
  if (type == WS_EVT_CONNECT) {
    if (ws.count() < MAX_WS_CLIENT) {
      client->setCloseClientOnQueueFull(false);
      Serial.printf("Client #%" PRIu32 " connected.\n", client->id());
    } else {
      client->text("{\"type\":\"redirect\",\"url\":\"/max_ws\"}");
      client->close(1008, "max client reached");
      Serial.printf("max client reached");
    }
  } else if (type == WS_EVT_DISCONNECT) {
    Serial.printf("Client #%" PRIu32 " disconnected.\n", client->id());
  } else if (type == WS_EVT_ERROR) {
    Serial.printf("Client #%" PRIu32 " error.\n", client->id());
  } else if (type == WS_EVT_DATA) {
    Serial.printf("Client #%" PRIu32 " len: %u\n", client->id(), len);
  }
}
```

but this causes also a crash with an interesting backtrace. This does not seem to prevent MAX_WS_CLIENT+1 from being opened. Although the connection is closed again, it is shortly open and then recognised by ws.cleanupClients().

```
Client #1 connected.
Client #2 connected.
max client reachedClient #1 len: 18
Client #1 len: 114
Client #1 len: 50
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 120
Client #1 len: 125
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 44
Client #1 len: 64
Client #1 len: 47
Client #1 len: 13
Client #1 len: 111
Client #1 len: 31
Client #1 error.
Client #1 len: 18
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 0
Client #1 len: 120
Client #1 len: 2
Client #1 len: 68
Client #1 len: 64
Client #1 len: 36
Client #1 len: 79
Client #1 len: 0
Client #1 len: 48
[... 225 lines cut ...]
  #2  0x40090784 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:251
      (inlined by) multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:240
  #3  0x40083526 in heap_caps_malloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps_base.c:128
  #4  0x40083261 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:84
  #5  0x4008328d in heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:110
  #6  0x400915b9 in malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c:24
  #7  0x400d4ed4 in AsyncWebSocketControl::AsyncWebSocketControl(unsigned char, unsigned char const*, unsigned int, bool) at .pio/libdeps/esp32/ESPAsyncWebServer/src/AsyncWebSocket.cpp:178
      (inlined by) AsyncWebSocketControl::AsyncWebSocketControl(unsigned char, unsigned char const*, unsigned int, bool) at .pio/libdeps/esp32/ESPAsyncWebServer/src/AsyncWebSocket.cpp:170
  #8  0x400d5c47 in decltype (::new ((void*)(0)) AsyncWebSocketControl((declval<unsigned char&>)(), (declval<unsigned char const*&>)(), (declval<unsigned int&>)(), (declval<bool&>)())) std::construct_at<AsyncWebSocketControl, unsigned char&, unsigned char const*&, unsigned int&, bool&>(AsyncWebSocketControl*, unsigned char&, unsigned char const*&, unsigned int&, bool&) at /Users/toe/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/stl_construct.h:97
      (inlined by) void std::allocator_traits<std::allocator<AsyncWebSocketControl> >::construct<AsyncWebSocketControl, unsigned char&, unsigned char const*&, unsigned int&, bool&>(std::allocator<AsyncWebSocketControl>&, AsyncWebSocketControl*, unsigned char&, unsigned char const*&, unsigned int&, bool&) at /Users/toe/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/alloc_traits.h:518
      (inlined by) AsyncWebSocketControl& std::deque<AsyncWebSocketControl, std::allocator<AsyncWebSocketControl> >::emplace_back<unsigned char&, unsigned char const*&, unsigned int&, bool&>(unsigned char&, unsigned char const*&, unsigned int&, bool&) at /Users/toe/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/deque.tcc:170
      (inlined by) AsyncWebSocketClient::_queueControl(unsigned char, unsigned char const*, unsigned int, bool) at .pio/libdeps/esp32/ESPAsyncWebServer/src/AsyncWebSocket.cpp:402
  #9  0x400d603e in AsyncWebSocketClient::_onData(void*, unsigned int) at .pio/libdeps/esp32/ESPAsyncWebServer/src/AsyncWebSocket.cpp:571
  #10 0x400d60c1 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(void*, AsyncClient*, void*, unsigned int)#5}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, void*&&, unsigned int&&) at .pio/libdeps/esp32/ESPAsyncWebServer/src/AsyncWebSocket.cpp:288
      (inlined by) __invoke_impl<void, AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::<lambda(void*, AsyncClient*, void*, size_t)>&, void*, AsyncClient*, void*, unsigned int> at /Users/toe/.platformio/packages/toolch [... cut ...]
  #12 0x4015e7a9 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp32/AsyncTCP/src/AsyncTCP.cpp:1400
  #13 0x4015f14a in _async_service_task(void*) at .pio/libdeps/esp32/AsyncTCP/src/AsyncTCP.cpp:190
      (inlined by) _async_service_task at .pio/libdeps/esp32/AsyncTCP/src/AsyncTCP.cpp:225
  #14 0x4008e676 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

```

mathieucarbou, 2024-11-29:
> I tried to prevent that more than MAX_WS_CLIENTS be opened with this code:

I am 100% sure than you should not `close()` within the context of an async_tcp callback which expects to be able to send a reply when returning.

There is a method available already to cleanup the clients: `_ws->cleanupClients(4);`

This can be executed from the main loop, each second for example.

mathieucarbou, 2024-11-29:
Note: if you want to prevent a WS connection from being opened, you can use middleware instead.
When the connection is upgraded, the initial request goes through the middleware chain.
So it is possible to write a rate limiter or middleware holding a semaphore and counting the clients and if too many, sends back an http status error code instead of calling `next()` in the middleware (which would go to the websocket handler upgrading the connection).

dewenni, 2024-11-29:
I have already used ws.cleanupClients to prevent several clients from remaining open unintentionally. But this can also lead to a crash, at least when using Safari. https://github.com/mathieucarbou/ESPAsyncWebServer/issues/162#issuecomment-2508582545. I think it is the same as the initial issue with ws.closeAll()

Unfortunately, I'm not really familiar with the topic of middleware. Would it be possible for you to make me a small example? For stability reasons, I would like to prevent more than ‘MAX_WS_CLIENTS’ from being opened. When trying to open MAX_WS_CLIENTS+1 I would like to redirect to another page.

mathieucarbou, 2024-11-30:
> Would it be possible for you to make me a small example?

Yes, quite easy to do you will see :-)

See the highlighted lines:

https://github.com/mathieucarbou/ESPAsyncWebServer/blob/main/examples/SimpleServer/SimpleServer.ino#L705-L721

The key is to understand the powerful concept of middleware and what they can do.

dewenni, 2024-12-08:
thanks @mathieucarbou it basically works as expected.
But I struggle to implement a redirect at this point to a different page.
I thought it could be simply handled like this:

```
  server.addHandler(&ws).addMiddleware([](AsyncWebServerRequest *request, ArMiddlewareNext next) {
    if (ws.count() >= MAX_WS_CLIENT) {
      // too many clients - answer back immediately and stop processing next middlewares and handler
      Serial.println("Redirecting due to too many clients.");
      request->redirect("/max_ws");
    } else {
      // process next middleware and at the end the handler
      next();
    }
  });
```

but it doesn´t work?! I tried also some other variants to implement this but none of them worked.
Could you give me a further advice?

mathieucarbou, 2024-12-08:
@dewenni : you cannot redirect on a websocket open request since this is the javascript `WebSocket` object that is handling the connection. What you can do though is to add a onError handler on the javascript object and redirect in javascript in case an error occurs: you should be able to introspect the error code I think. You need to look how the javascript API works. There is nothing to do on server-side: this is how the protocol works.

dewenni, 2024-12-08:
ok, I see.
but wouldn't it then be possible to use the middleware not at websocket level but directly at server level?

```c++
  server.addMiddleware([](AsyncWebServerRequest *request, ArMiddlewareNext next) {
    if (request->method() == HTTP_GET && ws.count() >= MAX_WS_CLIENT) {
      Serial.println("Too many WebSocket clients, redirecting client...");
      request->redirect("/max_ws");
    } else {
      next();
    }
  });
```

Here the condition also works as expected, but the redirect doesn't work either!
sorry for my lack of knowledge :-)

mathieucarbou, 2024-12-08:
@dewenni I think you didn't understand my answer: what you wrote leads to the same issues I've explained.
You need to do a UI change. Sending a redirect is not understood by the websocket JavaScript objet. You can send an http error code causing an error which you will catch in the websocket handler

dewenni, 2024-12-08:
My understanding is as follows.
If I define middleware in this way, I can do something between requesting a websocket connection and executing it.
```c++
server.addHandler(&ws).addMiddleware([](AsyncWebServerRequest *request, ArMiddlewareNext next)
..
```
In this case, however, I cannot make a redirect because the request is a web socket connection.

But I thought if I define it this way (without `addHandler(&ws)`), then I have the possibility to intercept an HTTP request and then, depending on what I put in there for a condition, I can also achieve a redirection. So maybe also check here in the condition how many websocket connections are already open.

```c++
server.addMiddleware([](AsyncWebServerRequest *request, ArMiddlewareNext next)
..
```

dewenni, 2024-12-08:
For completeness, your approach works  if I implement it like this, for example:

```c++
  server.addHandler(&ws).addMiddleware([](AsyncWebServerRequest *request, ArMiddlewareNext next) {
    if (ws.count() >= MAX_WS_CLIENT) {
      // too many clients - answer back immediately and stop processing next middlewares and handler
      Serial.println("Redirecting due to too many clients.");
      request->send(503, "text/plain", "Redirecting due to too many clients.");
    } else {
      // process next middleware and at the end the handler
      next();
    }
  });
```

in combination with this on the client side:

```js
  ws.onerror = function (error) {
    console.log("WebSocket error:", error);
    window.location.href = "/max_ws";
  };
```

I was just wondering if it is possible to do the redirection on the server side if I already react to the HTTP_GET request. But it works that way too.

mathieucarbou, 2024-12-10:
@dewenni : sorry yes you are right. My example was code written from memory but the mdidlewarre has to be applied ONLY to the websocket handler, not globally to the whole server like I did in my example!

`server.addHandler(&ws).addMiddleware([](AsyncWebServerRequest *request, ArMiddlewareNext next) {` is the correct way to go ;-)

> I was just wondering if it is possible to do the redirection on the server side if I already react to the HTTP_GET request. 

It would be possible to "test" by doing a `WeSocket(...).close()`, but if you do that, you need to introduce cookie support and sessions in order or keep some states about which http client tested and has the right to connect in the next page. This becomes more complex than just catching the error, showing a popuup or something tellng the user that the server is too busy and he should retry later.

pay attention that if you implement such mechanism, anyway, you need some sort of application-level inactivity timeout because someone could just leave his browser tab opened, go have lunch, and it will stall an opened connection on the server which will be inactive.

mathieucarbou, 2024-12-10:
Note about the Safari WebSocket issue, searching on Internet, there are many issues related to WebsSocket and Safari.
I do not know sadly enough of the websocket protocol to be able to fix that or update the code or even assert if there is something in the code that is impacted by Safari different behaviour.

mathieucarbou, 2024-12-10:
@dewenni @vortigont : Finally ! I was able to pinpoint the issue!

See PR: https://github.com/mathieucarbou/ESPAsyncWebServer/pull/168

mathieucarbou, 2024-12-10:
PR merged in main!

@dewenni : could you please point to `https://github.com/mathieucarbou/ESPAsyncWebServer` and test ? 

I will issue a new release once you have confirmed the fix.

It works for me on Safari.

dewenni, 2024-12-10:
thanks a lot - I will check as soon as possible and give Feedback

dewenni, 2024-12-10:
also the code from this example still crashes 
examples/Issue162/Issue162.ino

```
Client count: 0
Client count: 0
Client count: 0
Client count: 0
Client count: 0
Client count: 0
Client #1 connected.
Client count: 1
Closing all WebSocket clients...
Client #1 len: 0
Client #1 error.
Client #1 len: 68

assert failed: block_trim_free tlsf.c:496 (block_is_free(block) && "block must be free")

Backtrace: 0x40082325:0x3ffb3000 0x4008c151:0x3ffb3020 0x4009159a:0x3ffb3040 0x40090cb9:0x3ffb3170 0x4009064d:0x3ffb3190 0x40090784:0x3ffb31b0 0x40083526:0x3ffb31d0 0x40083261:0x3ffb3200 0x4008328d:0x3ffb3220 0x400915b9:0x3ffb3240 0x400fd80c:0x3ffb3260 0x400ed5b3:0x3ffb3280 0x400f1c97:0x3ffb32a0 0x400f1df6:0x3ffb32c0 0x400ee1c3:0x3ffb32e0 0x400ef0cc:0x3ffb3300 0x400ef0fb:0x3ffb3330 0x400ef111:0x3ffb3350 0x4015e26c:0x3ffb3370 0x4015e3c0:0x3ffb3390 0x4015e411:0x3ffb33b0 0x400f140e:0x3ffb33e0 0x400f5f0e:0x3ffb3420 0x400faabb:0x3ffb3450 0x400fd2a9:0x3ffb3470 0x4008e676:0x3ffb34a0
  #0  0x40082325 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:466
  #1  0x4008c151 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
  #2  0x4009159a in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:81
  #3  0x40090cb9 in block_locate_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/tlsf/tlsf.c:566
      (inlined by) tlsf_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/tlsf/tlsf.c:1005
  #4  0x4009064d in multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:207
  #5  0x40090784 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:251
      (inlined by) multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:240
  #6  0x40083526 in heap_caps_malloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps_base.c:128
  #7  0x40083261 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:84
  #8  0x4008328d in heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:110
  #9  0x400915b9 in malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c:24
  #10 0x400fd80c in mem_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/mem.c:209
  #11 0x400ed5b3 in pbuf_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c:284
  #12 0x400f1c97 in tcp_enqueue_flags at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_out.c:1077 (discriminator 2)
  #13 0x400f1df6 in tcp_send_fin at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_out.c:1020
  #14 0x400ee1c3 in tcp_close_shutdown_fin at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:431
  #15 0x400ef0cc in tcp_close_shutdown at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:403
  #16 0x400ef0fb in tcp_close_ext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:504 (discriminator 1)
  #17 0x400ef111 in tcp_close at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:486
  #18 0x4015e26c in AsyncClient::_lwip_fin(tcp_pcb*, signed char) at .pio/libdeps/esp32/AsyncTCP/src/AsyncTCP.cpp:987
  #19 0x4015e3c0 in AsyncClient::_s_lwip_fin(void*, tcp_pcb*, signed char) at .pio/libdeps/esp32/AsyncTCP/src/AsyncTCP.cpp:1408
  #20 0x4015e411 in _tcp_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp32/AsyncTCP/src/AsyncTCP.cpp:331
  #21 0x400f140e in tcp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:552 (discriminator 1)
  #22 0x400f5f0e in ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:752
  #23 0x400faabb in ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:186
  #24 0x400fd2a9 in tcpip_thread_handle_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:174
      (inlined by) tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:148
  #25 0x4008e676 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

```

mathieucarbou, 2024-12-10:
@dewenni : are you sure you are pointing to `https://github.com/mathieucarbou/ESPAsyncWebServer` ?

Or did you fetch upstream and pulled ?

- the new code prevents the callback to be called upon close, and they are called in your reports above.
- also, the new code has loop displaying the number of clients connected, and I do not see it in your report

To me, this is like you are running with old code still.

mathieucarbou, 2024-12-10:
@dewenni : ok, you can now point to branch [`issue-162`](https://github.com/mathieucarbou/ESPAsyncWebServer/tree/issue-162): `https://github.com/mathieucarbou/ESPAsyncWebServer#issue-162`

dewenni, 2024-12-10:
oh sorry I think I did a mistake by using:
lib_deps =
  mathieucarbou/AsyncTCP @ 3.2.14
  mathieucarbou/ESPAsyncWebServer
  
That will not use the latest commit of the main branch. It will use the latest release. Sorry.

mathieucarbou, 2024-12-10:
Ahah ;-) Yeah it will point to the latest from PIO registry :P You can point directly to the github repo by replacing:

`mathieucarbou/ESPAsyncWebServer`

by

`https://github.com/mathieucarbou/ESPAsyncWebServer#issue-162`

:-)

dewenni, 2024-12-10:
yes, I actually know that too, but somehow I hadn't thought about it.
Anyway, now I can confirm that both example codes works!
I can force some error messages, but no more crash.

**example with ws.CloseAll()**

```
Client count: 1
[ 52169][E][AsyncWebSocket.cpp:513] _onData(): WS[3]: _onData: 8
[ 52175][E][AsyncWebSocket.cpp:514] _onData(): WS[3]: _status = 1
[ 52181][E][AsyncWebSocket.cpp:515] _onData(): WS[3]: _pinfo: index: 0, final: 1, opcode: 8, masked: 1, len: 2
Client #3 disconnected.
Client #4 connected.
[ 53794][E][AsyncWebSocket.cpp:513] _onData(): WS[4]: _onData: 8
[ 53801][E][AsyncWebSocket.cpp:514] _onData(): WS[4]: _status = 1
[ 53807][E][AsyncWebSocket.cpp:515] _onData(): WS[4]: _pinfo: index: 0, final: 1, opcode: 8, masked: 1, len: 2
Client #4 disconnected.
Client #5 connected.
[ 54928][E][AsyncWebSocket.cpp:513] _onData(): WS[5]: _onData: 8
[ 54934][E][AsyncWebSocket.cpp:514] _onData(): WS[5]: _status = 1
[ 54940][E][AsyncWebSocket.cpp:515] _onData(): WS[5]: _pinfo: index: 0, final: 1, opcode: 8, masked: 1, len: 2
Client #5 disconnected.
Client #6 connected.
Client count: 1
Client count: 1
Closing all WebSocket clients...
[ 62143][E][AsyncWebSocket.cpp:513] _onData(): WS[6]: _onData: 2
[ 62149][E][AsyncWebSocket.cpp:514] _onData(): WS[6]: _status = 0
[ 62155][E][AsyncWebSocket.cpp:515] _onData(): WS[6]: _pinfo: index: 0, final: 1, opcode: 8, masked: 1, len: 0
[ 62165][E][AsyncWebSocket.cpp:513] _onData(): WS[6]: _onData: 4
[ 62171][E][AsyncWebSocket.cpp:514] _onData(): WS[6]: _status = 2
[ 62177][E][AsyncWebSocket.cpp:515] _onData(): WS[6]: _pinfo: index: 0, final: 0, opcode: 13, masked: 0, len: 113
Client #6 len: 2
Client #6 disconnected.
Client count: 0
Closing all WebSocket clients...
Client count: 0

```

**example with ws.cleanupClients(2)**

```
Client #1 connected.
Client #2 connected.
Client #3 connected.
Client #1 disconnected.
Client #4 connected.
Client #2 disconnected.
Client #5 connected.
[ 29311][E][AsyncWebSocket.cpp:513] _onData(): WS[3]: _onData: 2
[ 29318][E][AsyncWebSocket.cpp:514] _onData(): WS[3]: _status = 0
[ 29324][E][AsyncWebSocket.cpp:515] _onData(): WS[3]: _pinfo: index: 0, final: 1, opcode: 8, masked: 1, len: 0
[ 29334][E][AsyncWebSocket.cpp:513] _onData(): WS[3]: _onData: 4
[ 29339][E][AsyncWebSocket.cpp:514] _onData(): WS[3]: _status = 2
[ 29345][E][AsyncWebSocket.cpp:515] _onData(): WS[3]: _pinfo: index: 0, final: 0, opcode: 3, masked: 0, len: 117
Client #3 len: 2
Client #3 disconnected.

```

Thank you for keeping the issue alive and solving it. Thank you very much!.

mathieucarbou, 2024-12-10:
cool ! so it works: the logs you see are the ones I activated just now, just for this branch. But I will delete them and the branch.

You can point to main now in your PIO file:

`https://github.com/mathieucarbou/ESPAsyncWebServer`

The release will be in a few days...