mathieucarbou/ESPAsyncWebServer #172: [BUG] HTTPS request to HTTP crashes AsyncTCP or ESPAsyncWebServer

BeaverUI, 2024-12-15:
**Description**

Running PlatformIO with espressif32@6.9.0, mathieucarbou AsyncTCP and ESPAsyncWebServer. Custom settings:
  -D CONFIG_ASYNC_TCP_PRIORITY=10
  -D CONFIG_ASYNC_TCP_QUEUE_SIZE=256
  -D CONFIG_ASYNC_TCP_STACK_SIZE=8192
  -D CONFIG_ASYNC_TCP_RUNNING_CORE=1

When a request is made through HTTPS to the HTTP port, the web server (or AsyncTCP, perhaps) crashes fairly reliably, instead of just returning a request error. Sometimes it requires a few CTRL-F5 attempts for the core panic to occur.

**Stack trace**

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420d36b6  RA      : 0x420d3718  SP      : 0x3fca4450  GP      : 0x3fc8e200  
TP      : 0x3fc74f78  T0      : 0x00000006  T1      : 0xe6baac4d  T2      : 0x7201a8c0  
S0/FP   : 0x3fcb2d7c  S1      : 0x3fcb2d7c  A0      : 0x00000000  A1      : 0x3fcb2d7c  
A2      : 0x00000000  A3      : 0x3fcb2888  A4      : 0x00000000  A5      : 0x420d3710  
A6      : 0x00000000  A7      : 0x00000000  S2      : 0x3fc9a000  S3      : 0x3fc99386  
S4      : 0x3fc99380  S5      : 0x3fc99390  S6      : 0x3fc9a000  S7      : 0x3fc99384  
S8      : 0x3fc99388  S9      : 0x3fc94a38  S10     : 0x3fc993a6  S11     : 0x00000010  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x3fcb2d7c  T6      : 0x3fcb3cec  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000030  
MHARTID : 0x00000000  

Stack memory:
3fca4450: 0x3fc9a000 0x3fc9938c 0x3fcb2d7c 0x420d3718 0x3fc9a000 0x3fc9938c 0x3fcb2d7c 0x4205dac0
  #0  0x3fc9938c in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:85
  #1  0x420d3718 in AsyncServer::_s_accept(void*, tcp_pcb*, signed char) at lib/AsyncTCP/src/AsyncTCP.cpp:1591
  #2  0x3fc9938c in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:85
  #3  0x4205dac0 in tcp_process at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:947
      (inlined by) tcp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:438

3fca4470: 0x3fca4d6c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca4490: 0x00000000 0x00000000 0x3fcb2d7c 0x00000000 0x00000000 0x00000000 0x00000001 0x00000000
3fca44b0: 0x3fc9a000 0x00000014 0x3fca622c 0x3fca622c 0x3fcb27c8 0x3fc99ad8 0x3fcb4caa 0x42063776
  #0  0x3fc99ad8 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ip.c:64
  #1  0x42063776 in ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:804

3fca44d0: 0x00000004 0x00000003 0x00000000 0x00000000 0x00000004 0x00000003 0x3fcb2598 0x00000002
3fca44f0: 0x3fca622c 0x00000008 0x3fcb27c8 0x42067ee2 0x3fc99000 0xffffffff 0x3fcb4054 0x42056896
  #0  0x42067ee2 in ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:253
  #1  0x42056896 in tcpip_thread_handle_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:180
      (inlined by) tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:154

3fca4510: 0x00000000 0x00000000 0x00000000 0x3fcb4054 0x00000000 0x00000000 0x00000000 0x00000000
3fca4530: 0x00000000 0x00000000 0x00000000 0x403897b4 0x00000000 0x00000000 0x00000000 0x00000000
  #0  0x403897b4 in _prvTaskExitError at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/riscv/port.c:126
      (inlined by) prvTaskExitError at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/riscv/port.c:140

3fca4550: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678
3fca4570: 0x00000160 0xabba1234 0x00000154 0x3fca43f0 0x00004c09 0x3fc95750 0x3fc95750 0x3fca457c
3fca4590: 0x3fc95748 0x00000007 0x3fca4d74 0x3fca4d74 0x3fca457c 0x00000000 0x00000012 0x3fca396c
3fca45b0: 0x00546974 0x00000000 0x40100000 0x001ca50c 0x00000000 0x3fca4560 0x00000012 0x00000000
3fca45d0: 0x00000000 0x00000000 0x00000000 0x3fc9a914 0x3fc9a97c 0x3fc9a9e4 0x00000000 0x00000000
3fca45f0: 0x00000001 0x00000000 0x00000000 0x00000000 0x420d49ae 0x00000000 0x00000000 0x00000000
  #0  0x420d49ae in _cleanup_r at /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

3fca4610: 0x3fcb3998 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca4630: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca4650: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca4670: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca4690: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca46b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca46d0: 0xbaad5678 0x000000e0 0xabba1234 0x000000d4 0x3fca4734 0x3fca4754 0x3fca47b4 0x3fca4750
3fca46f0: 0x00000000 0x3fca46f8 0xffffffff 0x3fca46f8 0x3fca46f8 0x00000001 0x3fca470c 0xffffffff
3fca4710: 0x3fca341c 0x3fca341c 0x00000000 0x00000020 0x00000004 0x0000ffff 0x00000000 0xb33fffff
3fca4730: 0x00000000 0x3fcacf88 0x3fcad688 0x3fcad010 0x3fcac674 0x3fcacd4c 0x3fcadd00 0x3fcae0f8
3fca4750: 0x3fcaea30 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca4770: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca4790: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca47b0: 0xa5a5a5a5 0xbaad5678 0x00000060 0xabba1234 0x00000054 0x00000000 0x3fca47c4 0x00000000
3fca47d0: 0x00000000 0x00000000 0x3fca47dc 0xffffffff 0x3fca47dc 0x3fca47dc 0x00000000 0x3fca47f0
3fca47f0: 0xffffffff 0x3fca47f0 0x3fca47f0 0x00000001 0x00000001 0x00000000 0x0000ffff 0x00000000
3fca4810: 0xb33fffff 0x00000000 0xbaad5678 0x00000018 0xabba1234 0x0000000c 0x00000000 0x3fca4844
3fca4830: 0x00000000 0xbaad5678 0x0000001c 0xabba1234 0x00000010 0x3c1002e4 0x3fca4eec 0x3fca38d0

How can I avoid this? Some of my users are using port forwarding, and browsers attempt HTTPS first, resulting in problems when accessing if you just enter the web address without adding http:// explicitly. Apart from that, crashing because of such requests should never occur imho.

BeaverUI, 2024-12-15:
Another coredump error I sometimes get is this one:
``
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420d2dc6  RA      : 0x420d3128  SP      : 0x3fca3410  GP      : 0x3fc8e200  
TP      : 0x3fc73f58  T0      : 0x00000006  T1      : 0xb801a8c0  T2      : 0x7201a8c0  
S0/FP   : 0x00000000  S1      : 0x3fcb5a60  A0      : 0x00000000  A1      : 0x3fcb5a60  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3fc99494  A5      : 0x3fcb5b40  
A6      : 0x000013e7  A7      : 0x0000059c  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x3fc99380  S5      : 0x3fc99390  S6      : 0x3fc9a000  S7      : 0x3fc99384  
S8      : 0x3fc99388  S9      : 0x3fc94a38  S10     : 0x3fc993a6  S11     : 0x00000011  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x3fcb4608  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fca3410: 0x00000000 0x3fcb5a60 0x00000000 0x420d319c 0x3fca2868 0x00000000 0x3fc99388 0x3fcb5b40
  #0  0x420d319c in _tcp_recv(void*, tcp_pcb*, pbuf*, signed char) at lib/AsyncTCP/src/AsyncTCP.cpp:345
  #1  0x3fc99388 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:86

3fca3430: 0x3fc9a000 0x3fc99390 0x3fc99380 0x3fc99386 0x3fc9a000 0x3fc9938c 0x3fcb5a60 0x4205dd84
  #0  0x3fc99390 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:80
  #1  0x3fc99380 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:80
  #2  0x3fc99386 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:81
  #3  0x3fc9938c in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:85
  #4  0x4205dd84 in tcp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:542 (discriminator 1)

3fca3450: 0x3fca288c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3470: 0x00000000 0x00000000 0x3fcb5dc0 0x00000000 0x00000000 0x00000000 0x00000001 0x00000000
3fca3490: 0x3fc9a000 0x00000014 0x3fca5ed8 0x3fca5ed8 0x3fcb4470 0x3fc99ad8 0x3fcb746e 0x42063776
  #0  0x3fc99ad8 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ip.c:64
  #1  0x42063776 in ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:804

3fca34b0: 0x00000004 0x00000003 0x00000000 0x00000000 0x00000004 0x00000003 0x3fcb4238 0x00000002
3fca34d0: 0x3fca5ed8 0x00000008 0x3fcb4470 0x42067ee2 0x3fc99000 0xffffffff 0x3fcb47b0 0x42056896
  #0  0x42067ee2 in ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:253
  #1  0x42056896 in tcpip_thread_handle_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:180
      (inlined by) tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:154

3fca34f0: 0x00000000 0x00000000 0x00000000 0x3fcb47b0 0x00000000 0x00000000 0x00000000 0x00000000
3fca3510: 0x00000000 0x00000000 0x00000000 0x403897b4 0x00000000 0x00000000 0x00000000 0x00000000
  #0  0x403897b4 in _prvTaskExitError at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/riscv/port.c:126
      (inlined by) prvTaskExitError at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/riscv/port.c:140

3fca3530: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678
3fca3550: 0x00000160 0xabba1234 0x00000154 0x3fca3400 0x0000e34f 0x3fc95750 0x3fc95750 0x3fca355c
3fca3570: 0x3fc95748 0x00000007 0x3fca2894 0x3fca2894 0x3fca355c 0x00000000 0x00000012 0x3fca294c
3fca3590: 0x00546974 0x8ad41622 0x08c6e377 0x00b5bfe0 0x00000000 0x3fca3540 0x00000012 0x00000000
3fca35b0: 0x00000000 0x00000000 0x00000000 0x3fc9a914 0x3fc9a97c 0x3fc9a9e4 0x00000000 0x00000000
3fca35d0: 0x00000001 0x00000000 0x00000000 0x00000000 0x420d49ae 0x00000000 0x00000000 0x00000000
  #0  0x420d49ae in _cleanup_r at /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

3fca35f0: 0x3fcb5f10 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3610: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3630: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3650: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3670: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3690: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca36b0: 0xbaad5678 0x00000060 0xabba1234 0x00000054 0x3fca36c0 0x3fca36c0 0x3fca36c0 0x3fca36c0
3fca36d0: 0x00000000 0x3fca36d8 0xffffffff 0x3fca36d8 0x3fca36d8 0x00000000 0x3fca36ec 0xffffffff
[... cut ...]
3fca3750: 0xabba1234 0x000000d4 0x3fca37ac 0x3fca37cc 0x3fca382c 0x3fca37c8 0x00000000 0x3fca3770
3fca3770: 0xffffffff 0x3fca3770 0x3fca3770 0x00000001 0x3fca3784 0xffffffff 0x3fca576c 0x3fca576c
3fca3790: 0x00000000 0x00000020 0x00000004 0x0000ffff 0x00000000 0xb33fffff 0x00000000 0x3fcacc34
3fca37b0: 0x3fcad334 0x3fcaccbc 0x3fcac320 0x3fcac9f8 0x3fcad9ac 0x3fcadda4 0x3fcae6dc 0x00000001
3fca37d0: 0x00000000 0x001a7d88 0x3fca3fac 0x00000013 0x00000000 0x001baf88 0x3fca3fd0 0x00000004
3fca37f0: 0x00000000 0x001ce788 0x3fca4008 0x00000000 0x403b8000 0x001da08c 0x3fca4040 0x0000002d
``

mathieucarbou, 2024-12-15:
Hello,

Your stack size and queue size are huge: this is possible you run out of memory. 

Also, please do not post encoded stack traces like that: this is impossible to read and decrypt them.
Please follow the process described when opening an issue, or use PIO exception decoder.

There is nothing we can do so I will close. 

Please reopen:

1.  after you are using decoded stack trace
2. after having read it and fully made sure that the stack trace is not crashing because of your app or memory issues.

Thanks.

mathieucarbou, 2024-12-15:
>  #1  0x420d3718 in AsyncServer::_s_accept(void*, tcp_pcb*, signed char) at lib/AsyncTCP/src/AsyncTCP.cpp:1591

you do not seem to use latest versions. 

please make sure to upgrade also before opening an issue ;-)

BeaverUI, 2024-12-15:
Many thanks for your prompt reply, highly appreciated!

> Your stack size and queue size are huge: this is possible you run out of memory.

I did try reducing this before posting this issue, but it didn't influence the behaviour for HTTPS responses. Moreover, I need to transfer files of ~500 kB, and if I reduce the queue size and stack size, the ESP doesn't survive a stresstest (wdt triggers).

> Also, please do not post encoded stack traces like that: this is impossible to read and decrypt them. Please follow the process described when opening an issue, or use PIO exception decoder.

I'm using a debug build with the decoder enabled, hence the for the '#0  0x403897b4 in _prvTaskExitError at /home/runner/work/' messages. For the life of me I've never seen any more advanced debugging messages. I just did some Googling and as far as I can see I've configured things correctly. Could you please point me in the right direction? That would help me out immensely.

> There is nothing we can do so I will close.

While I understand you are very busy, and would like more advanced debugging data, I'm not sure this statement is true. I do see the behaviour in various implementations and am quite sure my code is OK since it runs fine otherwise.

> Note: we are all very busy and maintaining this OSS project on our free time so please make sure when you open an issue, there is inside enough info for us to do something ;-)

Your effort is highly appreciated! I hope you can point me in the direction of how to get more advanced exception decoding, so I can support as much as possible.

> > #1  0x420d3718 in AsyncServer::_s_accept(void*, tcp_pcb*, signed char) at lib/AsyncTCP/src/AsyncTCP.cpp:1591
> 
> you do not seem to use latest versions. please make sure to upgrade also before opening an issue ;-)
Thanks for the pointer, I recently pulled changes from the libs and assumed they were the latest versions. Will look into this as well.

mathieucarbou, 2024-12-15:
- AsyncTCP does not support HTTPS ;-)
- AsyncTCP supports a max of 16 opened sockets (defined by esp-idf compiler flag of Arduino Core)

- 500kB is not huge, but definitely slow if this is a file transfer. For this kind of usage, a workaround fix a has been implemented (see https://github.com/mathieucarbou/ESPAsyncWebServer/discussions/165#discussioncomment-11554123) and a more comprehensive fix is in progress and will be released early next week. This should solve the watchdog triggering on slow file reading

best thing you could do is:

1. upgrade
2. test the next release that will be done soon (this evening or tomorrow that will contain an AsyncTCP improvement  + https://github.com/mathieucarbou/ESPAsyncWebServer/pull/171)

to decode exceptions you can place that in your pio file:

`monitor_filters = esp32_exception_decoder`

Please re-open the issue once you have:

1. a proper stack trace
2. some code to show what you do ;)

thanks!

mathieucarbou, 2024-12-15:
@vortigont : FYI. 

this might be related to https://github.com/mathieucarbou/ESPAsyncWebServer/discussions/165 but not sure considering the little info we have....

BeaverUI, 2024-12-15:
I already had `monitor_filters = esp32_exception_decoder` in the ini, and build using debug. I honestly don't know what you're looking for; I think it can't decode some of the memory since it's from pre-built drivers/libs. 

After updating AsyncTCP to v3.3.0 and ESPAsyncWebServer to v3.4.1, the error did change a bit:

```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420d16c8  RA      : 0x420d172a  SP      : 0x3fca3470  GP      : 0x3fc8e200
TP      : 0x3fc73e9c  T0      : 0x00000006  T1      : 0xb801a8c0  T2      : 0x7201a8c0
S0/FP   : 0x3fcada60  S1      : 0x3fcada60  A0      : 0x00000000  A1      : 0x3fcada60
A2      : 0x00000000  A3      : 0x3fcb2320  A4      : 0x00000000  A5      : 0x420d1722
A6      : 0x00000000  A7      : 0x00000000  S2      : 0x3fc9a000  S3      : 0x3fc99386
S4      : 0x3fc99380  S5      : 0x3fc99390  S6      : 0x3fc9a000  S7      : 0x3fc99384
S8      : 0x3fc99388  S9      : 0x3fc94a38  S10     : 0x3fc993a6  S11     : 0x00000010
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x3fcada60  T6      : 0x3fcb23f8
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000030
MHARTID : 0x00000000

Stack memory:
3fca3470: 0x3fc9a000 0x3fc9938c 0x3fcada60 0x420d172a 0x3fc9a000 0x3fc9938c 0x3fcada60 0x4205dab8   
  #0  0x3fc9938c in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:85
  #1  0x420d172a in ieee80211_output_process at ??:?
  #2  0x3fc9938c in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:85
  #3  0x4205dab8 in lfs_bd_flush at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp_littlefs/src/littlefs/lfs.c:179

3fca3490: 0x3fca28cc 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000   
3fca34b0: 0x7201a8c0 0x00000006 0x00000000 0x00000000 0x00000000 0x00000000 0x00000001 0x00000000   
3fca34d0: 0x3fc9a000 0x00000014 0x3fca5f18 0x3fca5f18 0x3fcb4560 0x3fc99ad8 0x3fcb397e 0x4206376e   
  #0  0x3fc99ad8 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ip.c:64
  #1  0x4206376e in elf_process_tasks_regs at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/espcoredump/src/core_dump_elf.c:411
      (inlined by) elf_write_tasks_data at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/espcoredump/src/core_dump_elf.c:442
      (inlined by) esp_core_dump_do_write_elf_pass at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/espcoredump/src/core_dump_elf.c:551

3fca34f0: 0x00000004 0x00000003 0x00000000 0x00000000 0x00000004 0x00000003 0x3fcb2008 0x00000002   
3fca3510: 0x3fca5f18 0x00000008 0x3fcb4560 0x42067eda 0x3fc99000 0xffffffff 0x3fcadb74 0x4205688e   
  #0  0x42067eda in sswu at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/common/sae.c:692
  #1  0x4205688e in heap_caps_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps_init.c:98

3fca3530: 0x00000000 0x00000000 0x00000000 0x3fcadb74 0x00000000 0x00000000 0x00000000 0x00000000   
3fca3550: 0x00000000 0x00000000 0x00000000 0x403897b4 0x00000000 0x00000000 0x00000000 0x00000000   
  #0  0x403897b4 in _prvTaskExitError at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/riscv/port.c:126
      (inlined by) prvTaskExitError at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/riscv/port.c:140

3fca3570: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678   
3fca3590: 0x00000160 0xabba1234 0x00000154 0x3fca3440 0x002654bc 0x3fc95750 0x3fc95750 0x3fca359c   
3fca35b0: 0x3fc95748 0x00000007 0x3fca28d4 0x3fca28d4 0x3fca359c 0x00000000 0x00000012 0x3fca298c   
3fca35d0: 0x00546974 0x00000014 0xabba1234 0x00000008 0x00000000 0x3fca3580 0x00000012 0x00000000   
3fca35f0: 0x00000000 0x00000000 0x00000000 0x3fc9a914 0x3fc9a97c 0x3fc9a9e4 0x00000000 0x00000000   
3fca3610: 0x00000001 0x00000000 0x00000000 0x00000000 0x420d4c6c 0x00000000 0x00000000 0x00000000   
  #0  0x420d4c6c in loop_node_remove_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_event/esp_event.c:387
      (inlined by) esp_event_handler_unregister_with_internal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_event/esp_event.c:793

3fca3630: 0x3fcb2260 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000   
3fca3650: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000   
3fca3670: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000   
3fca3690: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000   
3fca36b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000   
3fca36d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000   
[... cut ...]
3fca37d0: 0x00000000 0x00000020 0x00000004 0x0000ffff 0x00000000 0xb33fffff 0x00000000 0x3fcacc74   
3fca37f0: 0x3fcad374 0x3fcaccfc 0x3fcac360 0x3fcaca38 0x3fcad90c 0x3fcadd5c 0x3fcae6d4 0x3fcb45cc   
3fca3810: 0x3fcb4540 0x001a0088 0x3fca3ff0 0x00000000 0x40318000 0x001b008c 0x3fca4014 0x00000004   
3fca3830: 0x00000000 0x001c0088 0x3fca404c 0x00000000 0x403b8000 0x001d008c 0x3fca4084 0x0000002d   
3fca3850: 0x00000000 0x001e0088 0x3fca40bc 0x0000002d 0x00000000 0x001f0088 0x3fca40f4 0xbaad5678

```

BeaverUI, 2024-12-15:
I'm happy to provide more information, but at the moment that doesn't seem possible (at least for me). However, triggering the problem is as easy as opening any webpage served by an ESP running your library through a browser using https://ESP:80 a few times in a row.

mathieucarbou, 2024-12-15:
I am wondering if it could be because you are on an old arduino version (espressif32@6.9.0).

could you try switching to latest arduino version ?

`platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10-rc3/platform-espressif32.zip`

Since Arduino 3, you need to user pioarduino since PlatformIO dropped support for Arduino.

BeaverUI, 2024-12-15:
I didn't know that support was dropped, will take a look to migrate things. I'm using an ESP32-C3-MINI chipset btw.

mathieucarbou, 2024-12-15:
Yes, sadly. The community is quite unhappy about that.

FYI ESP32-C3-MINI is single core, so you cannot have `CONFIG_ASYNC_TCP_RUNNING_CORE=1`.

ESPAsyncWS and ASyncTCP of course have more sense on dual-core.... If you are on single core, there is no point in using an asynchronous web server (although it should work).

You could directly use Arduino WebServer ;-)

Also, do you have this issue if you use other forks of AsyncTCP and ESPAsyncWS, or the original repos from me-no-dev ?
- me-no-dev/ESPAsyncWebServer
- me-no-dev/AsyncTCP

vortigont, 2024-12-16:
> a browser (Firefox in my case) using https://esp:80/ a few times in a row.

ugh that SSL-everywhere madness...
try to set `browser.fixup.fallback-to-https` to `false` in FF's `about:config`, it might help to remove such really stupid attempts to run SSL where it does not belongs to.

SSL is a binary protocol and can't be properly identified by current websever's code. We might try to improve it probably, but it would be an overhead code for a thing that should no be there anyway.

BeaverUI, 2024-12-16:
This morning I spent some time migrating everything to the latest platform (which required some code changes, but ala). Now, the good news is that the code runs like it used to. However, the bad news is that the code runs like it used to. It's perfectly stable until you do https://esp:80. This will crash it quite reliably.

mathieucarbou, 2024-12-16:
I already explained that AsyncTCP does not support SSL. Why are you still connecting with https to your esp ?

If you need SSL, use PsychicHttp, not ESPAsyncWS!

BeaverUI, 2024-12-16:
I believe there is a misunderstanding. My intention is NOT to use SSL. The intention is to avoid crashing when my users attempt to (accidentally) do so :)

vortigont, 2024-12-16:
BTW, I can't reproduce it with curl, it just closes the connection with 404, no crash.
like this:

```
* Connection from 192.168.1.100:46664
> ANY  HTTP/1.1
> �:
>
* Processed in 0 ms
< HTTP/1.1 404 Not Found
< connection: close
```

So it's not a 100% case and might depend on how webserver is set for your project.
I think you might want to start checking with a very basic example how it works and then compare on how handlers are set in your project. It might make a difference.

BeaverUI, 2024-12-16:
curl also doesn't cause a crash here, please test with Firefox instead.

mathieucarbou, 2024-12-16:
Tried to run the SimpleServer sample in the project: `https://192.168.4.1:80/` gives an ssl error in the browser and does not crash the esp, in safari, chrome, firefox on my Mac.

BeaverUI, 2024-12-17:
I did some more testing:
- It seems only Firefox causes the issue. Chromium and curl (both on Linux) don´t trigger the issue.
- Most of the time I need to CTRL-F5 a few times (or keep F5 pressed) to trigger it.
- Updating the ESP platform from 6.9.0 to 52.03.10 didn't change the behavior
- Updating ESPAsync and AsyncWS didn't solve it
- I completely removed almost all handles from the code, leaving only the initialization. It still happens with this piece of code:
```
AsyncWebServer web_server(80);

void InitWebserver(void){
  web_server
    .on("/fwupdate", HTTP_GET, [](AsyncWebServerRequest *request){
PROGMEM const char * text = R"(
<html>
<head></head>
<body>
test
</body>
</html>
)";
      request->send(200, "text/html", text);
    });

  web_server.begin();
}

With the latest libraries, my coredump is as follows:
```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420d9470  RA      : 0x420d94d4  SP      : 0x3fca7130  GP      : 0x3fc8f600  
TP      : 0x3fca7250  T0      : 0x00000001  T1      : 0x3fcaf144  T2      : 0x00000006  
S0/FP   : 0x3fcbdd18  S1      : 0x3fcbdd18  A0      : 0x00000000  A1      : 0x3fcbdd18  
A2      : 0x00000000  A3      : 0x3fcb7f14  A4      : 0x00000000  A5      : 0x420d94cc  
A6      : 0x00000000  A7      : 0x00000006  S2      : 0x3fc9a000  S3      : 0x3fc9a000  
S4      : 0x3fc9a000  S5      : 0x3fc9a000  S6      : 0x3fc96000  S7      : 0x3fc9691c  
S8      : 0x3fc9a000  S9      : 0x3fc9a000  S10     : 0x00000010  S11     : 0x3fc9a000  
T3      : 0xb801a8c0  T4      : 0x3fcbdd18  T5      : 0x3fcb8060  T6      : 0x0000000a  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000030  
MHARTID : 0x00000000  

Stack memory:
3fca7130: 0x3fc96000 0x3fc9a000 0x3fcbdd18 0x420d94d4 0x3fc9a000 0x3fc9a000 0x3fcbdd18 0x4205642c
  #0  0x3fc9a000 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:92
  #1  0x420d94d4 in AsyncServer::_s_accept(void*, tcp_pcb*, signed char) at .pio/libdeps/debug/AsyncTCP/src/AsyncTCP.cpp:1657
  #2  0x3fc9a000 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:92
  #3  0x3fc9a000 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:92
  #4  0x4205642c in tcp_process at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:956
      (inlined by) tcp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:449

3fca7150: 0x3fc9a000 0x00000001 0x3fc9a000 0x00000000 0x3fca71fc 0x00000000 0x3fca4f88 0x00000000
  #0  0x3fc9a000 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:92
  #1  0x3fc9a000 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:92

3fca7170: 0x00000000 0x00000000 0x00000000 0x00000001 0x00000000 0x00000014 0x3fcaf144 0x3fcaf144
3fca7190: 0x3fcb9762 0x3fc96730 0x3fcb8370 0x4205c586 0xffffffff 0x3fc9a000 0x3fcb86cc 0x00000000
  #0  0x3fc96730 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ip.c:64
  #1  0x4205c586 in ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:753
  #2  0x3fc9a000 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:92

3fca71b0: 0x00000000 0x00000000 0x00000000 0x3fc9a000 0x00000008 0x3fcaf144 0x3fcb8370 0x4206219a
  #0  0x3fc9a000 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:92
  #1  0x4206219a in ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:253

3fca71d0: 0xffffffff 0x3fc9a000 0x3fcaf44c 0x4204e606 0x00000000 0x00000000 0x00000000 0x0000005d
  #0  0x3fc9a000 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:92
  #1  0x4204e606 in tcpip_thread_handle_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:174 (discriminator 1)
      (inlined by) tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:148 (discriminator 1)

[... 7 lines cut ...]
3fca7290: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca72b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca72d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca72f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca7310: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca7330: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca7350: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[... cut ...]
3fca7490: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca74b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca74d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca74f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca7510: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: fbcfd9a5e
Rebooting...
```

mathieucarbou, 2024-12-17:
I was finally able to reproduce with FF, hitting refresh a couple f time with `https://192.168.4.1:80/`.

I think we could add a feature inside `AsyncWebServerRequest::_onData(void* buf, size_t len)` to detect ssl traffic and avoid crashing the ESP... I'll dig into it...

vortigont, 2024-12-17:
this SSL could be tricky. Now they try to enforce encrypted SSL hello's everywhere which would make it even more fun.

BeaverUI, 2024-12-17:
> I was finally able to reproduce with FF, hitting refresh a couple f time with `https://192.168.4.1:80/`.
> 
> I think we could add a feature inside `AsyncWebServerRequest::_onData(void* buf, size_t len)` to detect ssl traffic and avoid crashing the ESP... I'll dig into it...

I think it would be better to not attempt to detect SSL, but set checks and balances (e.g. max header packet length, detect certain packet elements) such that only plain HTTP traffic is processed. Reading through the _onData, there does not seem to be such checks. It seems to just hand over data to _parseLine and _parseReqHead to get the packet head. I would say that in _parseLine you need an extra function e.g. _checkReqHead, which checks whether it is indeed an HTTP header.

mathieucarbou, 2024-12-17:
> I think it would be better to not attempt to detect SSL, but set checks and balances (e.g. max header packet length, detect certain packet elements) such that only plain HTTP traffic is processed. [...]

Why would I complexity a solution that is really simple to implement ?

[image]

BeaverUI, 2024-12-17:
You're fast, I see you already pushed an update. Amazing work :+1:

mathieucarbou, 2024-12-17:
> You're fast, I see you already pushed an update. Amazing work 👍

Sorry for my previous misleading comments - I really failed to understand what you really wanted. Re-reading now makes more sense. I was convicted you were not really trying to reach the esp through an ssl connection at first, and thus that it was another issue 🤣

mathieucarbou, 2024-12-17:
v3.4.3 released!

BeaverUI, 2024-12-17:
No worries, it's great you picked it up. There is a memory leak now though :-P

mathieucarbou, 2024-12-17:
I am wondering if this is the created request object which is never sent... I will have a look.

mathieucarbou, 2024-12-17:
Ok so this is not caused by AsyncTCP changes. The request object is correctly deleted:

**https://192.168.4.1:80/** (1 hit creates 2 requests)

```
[2214841][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd4368
[2214848][D][WebRequest.cpp:71] _onData(): Free Heap: 214964
[2214855][D][WebRequest.cpp:77] _onData(): SSL/TLS handshake detected: closing connection
[2214866][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2214873][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd4368
[2214882][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd4234
[2214890][D][WebRequest.cpp:71] _onData(): Free Heap: 215592
[2214896][D][WebRequest.cpp:77] _onData(): SSL/TLS handshake detected: closing connection
[2214906][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2214913][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd4234
[2215330][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd4918
[2215337][D][WebRequest.cpp:71] _onData(): Free Heap: 213808
[2215344][D][WebRequest.cpp:77] _onData(): SSL/TLS handshake detected: closing connection
[2215354][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2215361][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd4918
[2215370][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd4234
[2215378][D][WebRequest.cpp:71] _onData(): Free Heap: 214668
[2215384][D][WebRequest.cpp:77] _onData(): SSL/TLS handshake detected: closing connection
[2215394][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2215401][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd4234
[2216144][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd563c
[2216152][D][WebRequest.cpp:71] _onData(): Free Heap: 212884
[2216158][D][WebRequest.cpp:77] _onData(): SSL/TLS handshake detected: closing connection
[2216169][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2216176][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd563c
[2216185][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd4234
[2216193][D][WebRequest.cpp:71] _onData(): Free Heap: 213520
[2216199][D][WebRequest.cpp:77] _onData(): SSL/TLS handshake detected: closing connection
[2216209][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2216216][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd4234
```

**http://192.168.4.1/** (2 hits)

```
[2084760][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd4150
[2084767][D][WebRequest.cpp:71] _onData(): Free Heap: 216664
[2084809][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2084817][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd4150
[2089123][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd411c
[2089130][D][WebRequest.cpp:71] _onData(): Free Heap: 216448
[2089163][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2089171][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd411c
[2184575][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd42c0
[2184583][D][WebRequest.cpp:71] _onData(): Free Heap: 216808
[2184611][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2184618][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd42c0
[2188501][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd42c0
[2188508][D][WebRequest.cpp:71] _onData(): Free Heap: 216808
[2188531][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2188538][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd42c0
[2190425][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd42c0
[2190433][D][WebRequest.cpp:71] _onData(): Free Heap: 216808
[2190453][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2190461][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd42c0
[2191723][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd42c0
[2191731][D][WebRequest.cpp:71] _onData(): Free Heap: 216808
[2191764][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2191772][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd42c0
[2192965][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd42c0
[2192972][D][WebRequest.cpp:71] _onData(): Free Heap: 216808
[2193002][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2193010][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd42c0
[2194146][D][WebRequest.cpp:70] _onData(): Request pointer: 0x3ffd42c0
[2194153][D][WebRequest.cpp:71] _onData(): Free Heap: 216808
[2194180][D][WebRequest.cpp:221] _onDisconnect(): Client disconnected
[2194187][D][WebServer.cpp:138] _handleDisconnect(): Deleting request pointer: 0x3ffd42c0```

mathieucarbou, 2024-12-17:
@vortigont : fyi I also checked line 1104 of AsyncTCP `pbuf_free` is correclty freeing the buffer.

I suspect the issue to be before ESPAsyncWS because the data from a new client arrives, a new request is created (`AsyncWebServerRequest* r = new AsyncWebServerRequest((AsyncWebServer*)s, c);`), then onData is called, the new code intercepts the SSL/TLS handshake and close the client and returns. Closing the client causes the disconnect handler to be called, which closes the client, deletes the request, and the client.

All that within the call of `_recv_cb(_recv_cb_arg, this, b->payload, b->len);` in AsyncTCP line 1095.

Then, `pbuf_free` is called.

vortigont, 2024-12-18:
it's not that scary actually
for me it releases after some time, but maybe loosing just a dozen of bytes or so...
interval between `heap:` messages is 3 sec

```
heap:263976
heap:263976
heap:263976
heap:263976
heap:263976
[504805][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[506235][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
heap:263496
[507259][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[508091][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[509109][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[510127][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
heap:262600
[511357][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[512381][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
heap:262152
[513812][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[514839][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[515656][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
heap:261480
[516683][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[517909][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[518935][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
heap:261704
[519753][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[520572][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[521597][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
heap:261928
heap:262600
heap:263272
[530813][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
heap:263272
[532451][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[533069][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[533683][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[534103][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
heap:263496
[534704][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[535318][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
[535933][D][WebRequest.cpp:76] _onData(): SSL/TLS handshake detected: closing connection
heap:263496
heap:263496
heap:263496
heap:263976
heap:263976
heap:263976
heap:263976
```

mathieucarbou, 2024-12-18:
I am wondering if it is browser dependent. For exemple, Firefox seems to retry. Also, even if globally, issuing several refresh does decrease the heap, sometimes, I see the feee heap value that is going up again. But not to the point where it was at start.

Searching on Google, I saw some older issues regarding SO_LINGER not properly implemented: set to 0 by default, which should make the esp-idf layer discard the packets when connection closes, but those issues were reporting that the code is not cleaning, and users had to activate SO_LINGER and set the cleanup time to 0 in order to force an immediate buffer release once the connection closes.

But whether this issue is still there or not, we do not have control over that. on Arduino Core.

Strangely, esp-idf http server seems to have control over that: 

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/esp_http_server.html?highlight=linger#_CPPv4N12httpd_config16enable_so_lingerE

mathieucarbou, 2024-12-18:
Out of curiosity, I tested with AsyncTCPSock, which is based on `lwip/sockets`, like the esp-idf implementation of http server.

Hitting refresh consecutively also does decreases the heap.

So I added this to activate so_linger with a 0 delay

```
        struct linger so_linger = {
            .l_onoff = true,
            .l_linger = 0,
        };
        setsockopt(accepted_sockfd, SOL_SOCKET, SO_LINGER, &so_linger, sizeof(struct linger));
```

And the free heap still decreases. But a little less than with AsyncTCP.

So its not that.

vortigont, 2024-12-18:
my 5 cents that it's lwip's internal allocations and garbage collection. It should keep tcp struct lingering for some time by the standard.

It would be much easy to use  `_client->abort();` instead of `_client->close();`. It will simply send TCP RST and  release all related structs instead of close via fin/ack sequence.

closed 2024-12-19 by mathieucarbou

mathieucarbou, 2024-12-19:
@BeaverUI : the memory leak is fixed in [v3.4.4](https://github.com/mathieucarbou/ESPAsyncWebServer/actions/runs/12410312309)!

BeaverUI, 2024-12-19:
many thanks, works perfectly