mathieucarbou/ESPAsyncWebServer #176: Websocket com not correctly working

mathieucarbou, 2024-12-18:
I have a similar issue after switching from esphome/ESPAsyncWebServer-esphome.
The issues is that `socket.textAll()` doesn't seem to work in a socket event handler.

Code looks like this:

```c++
AsyncWebSocket websocket("/ws");
int clients_connected = 0;
String test_msg = "A new client connected!";

// Socket event handler
void onSocketEvent(AsyncWebSocket* server, AsyncWebSocketClient* client, AwsEventType type, void* arg, uint8_t* data, size_t len) {
  switch (type) {
    case WS_EVT_CONNECT:
      ++clients_connected;

      websocket.textAll(test_msg);   // This message is not received by clients.

      // This is being executed correctly when a new client connects
      Serial.print("[Websocket] Client ");
      Serial.print(client->id());
      Serial.print(" connected with IP: ");
      Serial.println(client->remoteIP());
      break;

    case WS_EVT_DISCONNECT:
        --clients_connected;
     .
     .
     .
  }
}

// Called in setup()
void initWebsocket() {
  websocket.onEvent(onSocketEvent);
}
```

With the esphome branch of ESPAsyncWebserver this worked fine with multiple clients connected. The message/s sent are also very small.  It is only in the event handler that `websocket.textAll()` doesn't work. In the loop() it works just fine.

I tried both AsyncTCP and AsyncTCPSock.

_Originally posted by @stefan-wr in https://github.com/mathieucarbou/ESPAsyncWebServer/discussions/65#discussioncomment-11611026_

stefan-wr, 2024-12-19:
I tested a few older versions down to
````
mathieucarbou/Async TCP @ 3.0.0
mathieucarbou/ESP Async WebServer @ 2.4.1
````
The issues persists.

But I found out that it only affects the new socket client that just connected and caused the WS_EVT_CONNECTED event type.
All other previously connected clients receive the message. For example:
Client 1 and Client 2 are connected to websocket.
Now a third client, Client 3 connects and causes the `text_msg` to be sent from inside the event handler to all clients, but it is only received by Client 1 & 2.

If I send the message directly to the client from within the event handler, e.g.
```
client->text(test_msg);
```
The message reaches the client.

I dug a bit deeper into the `textAll()` function and added some logging. It seems that the new client doesn't make it into the list of clients `_clients` in time to be selected in the for loop.

```
void AsyncWebSocket::textAll(std::shared_ptr<std::vector<uint8_t>> buffer)
{   
    for (auto &c : _clients) {
        if (c.status() == WS_CONNECTED)
            c.text(buffer);
    }
}
```

vortigont, 2024-12-19:
It is by design.

```
AsyncWebSocketClient* AsyncWebSocket::_newClient(AsyncWebServerRequest* request) {
  _clients.emplace_back(request, this);
  return &_clients.back();
}
```

`WS_EVT_CONNECT` callback is called from the `AsyncWebSocketClient`'s constructor (which is probably wrong place for user code's callbacks) when object is not yet linked to a list of server's clients.

commit 2a26a97e (2024-12-19): move AsyncWebSocket's WS_EVT_CONNECT callback out of AsyncWebSocketClient's constructor

 - this is wrong place to call user code from an object constructor
 - it is wrong to call AsyncWebSocket's method from other's objects constructor
 - the callback is executed when new object is not yet linked to server's clients list

Closes #176
closed 2024-12-19 by mathieucarbou