mathieucarbou/ESPAsyncWebServer #182: [BUG]undefined reference to `AsyncWebServer::AsyncWebServer(unsigned short)'

mkeyno, 2024-12-21:
**Description**

I used the Me_No_dev asynwebserver lib in my program without any problem, however last week I tried to use this library instead, 
but whenever I switch to this library it makes the following error
any help appreciated 

```cpp
#include <Arduino.h>
#include "ArduinoJson.h"
#include "AsyncJson.h"
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <FS.h>
#include <LITTLEFS.h>

AsyncWebServer server(80);

class test_p
{
private:
 AsyncWebServer *Server;
public:
  test_p();
  void begin(AsyncWebServer *server);
  ~test_p();
};

test_p::test_p(){}
void test_p::begin(AsyncWebServer *server){
Server = server;

}
test_p::~test_p(){}

test_p asw_test;

void setup() {

  asw_test.begin(&server);
 
}

void loop() {}
```

error 

```
c:/users/lkeyn/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.cpp.o:(.literal.startup._GLOBAL__sub_I_server+0x0): undefined reference to `AsyncWebServer::AsyncWebServer(unsigned short)'
c:/users/lkeyn/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.cpp.o:(.literal.exit._GLOBAL__sub_D_server+0x0): undefined reference to `AsyncWebServer::~AsyncWebServer()'
c:/users/lkeyn/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.cpp.o: in function `_GLOBAL__sub_I_server':
C:\Users\lkeyn\Documents\PlatformIO\Projects\esp32_v01/src/main.cpp:10: undefined reference to `AsyncWebServer::AsyncWebServer(unsigned short)'
c:/users/lkeyn/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.cpp.o: in function `_GLOBAL__sub_D_server':
C:\Users\lkeyn\Documents\PlatformIO\Projects\esp32_v01/src/main.cpp:10: undefined reference to `AsyncWebServer::~AsyncWebServer()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1

```

proddy, 2024-12-21:
Not a bug. Your platformio configuration is not set up correctly. You are including the source and header files but pio is not compiling the AsyncWebServer sources so the linking breaks. You can explicitly add `build_src` to the platformio.ini (see pio's documentation) but it'll be much easier if you just remove the `build_flags` you set and follow the example described in this library at https://github.com/mathieucarbou/ESPAsyncWebServer?tab=readme-ov-file#dependencies

mathieucarbou, 2024-12-21:
@mkeyno : also make sure to use the lastest version of Arduino Core with pioarduino.

PlatformIO dropped support for Arduino Core so you cannot use `platform = espressif32` anymore. This pints to an old Arduino Core version...