mathieucarbou/ESPAsyncWebServer #188: libCompatMode prevents use with Arduino as an ESP-IDF component

thorrak, 2025-01-07:
As you know, Platformio has stopped directly supporting the latest Arduino framework releases for ESP32. As a result, I have begun to change my projects to use `framework=espidf` for Platformio with Arduino as a component.

Thus far I've had success using almost every library without any code changes, but Platformio's built in library finder balks when searching for dependencies unless I set `lib_compat_mode=off` within my `platformio.ini`. This again works for all of the libraries -- except this one and [AsyncTCP](https://github.com/mathieucarbou/AsyncTCP) which themselves specify a requirement for strict dependency checking.

I can confirm that both of these libraries work when using Arduino as a component for ESP-IDF, but to get them to properly download I have to [remove](https://github.com/thorrak/ESPAsyncWebServer/commit/9423a17de1e35c14e64e9d6d17079804a82414d8) libCompatMode from library.json.

From looking through the git blame, I can see where [this commit](https://github.com/mathieucarbou/ESPAsyncWebServer/commit/005c3f3b712e318887aea478d55024d98f5b2ae2) introduced the change. I can see a discussion of this change [here](https://github.com/mathieucarbou/ESPAsyncWebServer/pull/39#issuecomment-2195809788) as well. My reading of these makes it sound like the reason this change was introduced was to enhance CI -- would it be possible to use environment-specific `lib_compat_mode = strict` in CI rather than at the library level in order to retain compatibility with Arduino as a Component for ESP-IDF?

mathieucarbou, 2025-01-07:
> As you know, Platformio has stopped directly supporting the latest Arduino framework releases for ESP32. As a result, I have begun to change my projects to use `framework=espidf` for Platformio with Arduino as a component.

You could have used pioarduino: 

https://github.com/pioarduino/platform-espressif32

it is a platformio fork, community maintained, and we even have a vscode extension. Jason from Tasmota is the lead of this effort.

mathieucarbou, 2025-01-07:
I introduced https://github.com/mathieucarbou/ESPAsyncWebServer/commit/005c3f3b712e318887aea478d55024d98f5b2ae2 because platformio was bringing in consumer projects all the dependencies including the ones for 8266 and RPI.
I merge your PR and verify if adding `ib_compat_mode = strict` really solves it. From memory I still had issues but I do not remember if it was with the combination below that everyone should IMHO use.
```
lib_compat_mode = strict
lib_ldf_mode = chain
```

thorrak, 2025-01-07:
> You could have used pioarduino:
>
> https://github.com/pioarduino/platform-espressif32
>
> it is a platformio fork, community maintained, and we even have a vscode extension. Jason from Tasmota is the [...]

I initially did use pioarduino. Very glad to see it out there given Espressif's stance on both the ESP32 and RP2040! I ended up using Arduino as a Component as - for now - I wanted to retain the ability to use an existing platformio toolchain and separately needed the ability to reconfigure my project using menuconfig to enhance heap poisoning support.

mathieucarbou, 2025-01-07:
3.5.0 released ;-)

thorrak, 2025-01-07:
Cheers - thank you!

vortigont, 2025-01-08:
I've tried to use "Arduino as component"  with Platformio but faced many issues in such setups.
@thorrak could you pls share your experience in this area?
Which Arduino core version you have used, any working platformio configs you can share? Or do you have any public repos that I can check?
Thanks!

thorrak, 2025-01-08:
> I've tried to use "Arduino as component" with Platformio but faced many issues in such setups. @thorrak could you pls share your experience in this area? Which Arduino core version you have used, any working platformio configs you can share? Or do you have any public repos that I can check? Thanks!

I actually wrote something this afternoon documenting exactly what I did. Here’s the post - if you run into any issues or find I missed something, please let me know!

https://blog.fermentrack.com/converting-to-arduino-as-an-esp-idf-component/

mathieucarbou, 2025-01-08:
@thorrak : fyi, pioarduino suppots hybrid compilation.

Example: https://github.com/mathieucarbou/YaSolR/blob/main/platformio.ini#L156-L188

There are also other examples of that in the pioarduino repo.

You can remove esp-idf components (not add some), and you can also control all the esp-idf config. It will recompile the esp-idf code with Arduino and the new generated sdkconfig.

mathieucarbou, 2025-01-08:
> That’s awesome. Is that new for pioarduino? I recall searching for that eons ago, but never finding a way to do it.

Yes, this was in Jason personal branch but was merged since Arduino Core 3.1 in pioarduino. So it is available with: Arduino Core 3.1 ÷ ESP-IDF 5.3

`platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip`

> One thing I noticed in that repo you linked as well - Have you managed to get your [GitHub Actions build scripts](https://github.com/mathieucarbou/YaSolR/blob/main/.github/workflows/build.yml) working with pioarduino? This was one of the other issues I ran into when I had initially converted.

Yes, without any issue. You can have a look how I setup CI in all the libraries I do. YaSolR is a project so a little more complex than a library, but everything I do uses pioarduino. I even use its vscode extension.

https://github.com/mathieucarbou