mathieucarbou/ESPAsyncWebServer #191: [BUG] Support for RP2040 core has been broken with latest update

ayushsharma82, 2025-01-09:
Hi @mathieucarbou ,

It looks like support for RP2040/pico core has been broken with latest update. These are the CI errors of NetWizard dev branch:

```
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp: In member function 'bool AsyncWebSocket::availableForWriteAll()':
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp:77[6](https://github.com/ayushsharma82/NetWizard/actions/runs/12698528246/job/35397212592#step:15:7):15: error: 'none_of' is not a member of 'std'
  776 |   return std::none_of(std::begin(_clients), std::end(_clients), [](const AsyncWebSocketClient& c) { return c.queueIsFull(); });
      |               ^~~~~~~
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp: In member function 'bool AsyncWebSocket::availableForWrite(uint32_t)':
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp:[7](https://github.com/ayushsharma82/NetWizard/actions/runs/12698528246/job/35397212592#step:15:8)80:26: error: 'find_if' is not a member of 'std'
  780 |   const auto iter = std::find_if(std::begin(_clients), std::end(_clients), [id](const AsyncWebSocketClient& c) { return c.id() == id; });
      |                          ^~~~~~~
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp: In member function 'size_t AsyncWebSocket::count() const':
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp:7[8](https://github.com/ayushsharma82/NetWizard/actions/runs/12698528246/job/35397212592#step:15:9)7:15: error: 'count_if' is not a member of 'std'
  787 |   return std::count_if(std::begin(_clients), std::end(_clients), [](const AsyncWebSocketClient& c) { return c.status() == WS_CONNECTED; });
      |               ^~~~~~~~
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp: In member function 'AsyncWebSocketClient* AsyncWebSocket::client(uint32_t)':
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp:7[9](https://github.com/ayushsharma82/NetWizard/actions/runs/12698528246/job/35397212592#step:15:10)1:26: error: 'find_if' is not a member of 'std'
  791 |   const auto iter = std::find_if(_clients.begin(), _clients.end(), [id](const AsyncWebSocketClient& c) { return c.id() == id && c.status() == WS_CONNECTED; });
      |                          ^~~~~~~
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp: In member function 'void AsyncWebServerRequest::_removeNotInterestingHeaders()':
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp:156:[12](https://github.com/ayushsharma82/NetWizard/actions/runs/12698528246/job/35397212592#step:15:13): error: 'any_of' is not a member of 'std'
  156 |   if (std::any_of(std::begin(_interestingHeaders), std::end(_interestingHeaders), [](const String& str) { return str.equalsIgnoreCase(T_ANY); }))
      |            ^~~~~~
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp:162:[14](https://github.com/ayushsharma82/NetWizard/actions/runs/12698528246/job/35397212592#step:15:15): error: 'none_of' is not a member of 'std'
  162 |     if (std::none_of(std::begin(_interestingHeaders), std::end(_interestingHeaders), [&name](const String& str) { return str.equalsIgnoreCase(name); }))
      |              ^~~~~~~
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp: In member function 'const AsyncWebHeader* AsyncWebServerRequest::getHeader(const char*) const':
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp:626:20: error: 'find_if' is not a member of 'std'
  626 |   auto iter = std::find_if(std::begin(_headers), std::end(_headers), [&name](const AsyncWebHeader& header) { return header.name().equalsIgnoreCase(name); });
      |                    ^~~~~~~
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp: In member function 'void AsyncWebServerRequest::addInterestingHeader(const char*)':
/home/runner/Arduino/libraries/ESPAsyncWebServer/src/WebRequest.cpp:688:12: error: 'none_of' is not a member of 'std'
  688 |   if (std::none_of(std::begin(_interestingHeaders), std::end(_interestingHeaders), [&name](const String& str) { return str.equalsIgnoreCase(name); }))
      |            ^~~~~~~

Error during build: exit status 1
```

mathieucarbou, 2025-01-09:
@ayushsharma82 Ok, we'll look.

In the meantime, would you please make sure you have the latest version and also apply the tweaks in platformio.ini file to correctly fond the dependencies:

See the release notes: https://github.com/mathieucarbou/ESPAsyncWebServer/releases

ThankS!

ayushsharma82, 2025-01-09:
This is related to Arduino-CLI CI test: https://github.com/ayushsharma82/NetWizard/actions/runs/12698528246/job/35397212592

ayushsharma82, 2025-01-09:
Couple days of break from Github and totally forgot that we were using tagged library releases in CI yml haha. Just saw that CI was using `3.1.1` despite of `3.6.0` set in library.json which was the cause of this issue.

closed 2025-01-09 by ayushsharma82

ayushsharma82, 2025-01-09:
@mathieucarbou  Do you have any idea on why the following error is happening: https://github.com/ayushsharma82/NetWizard/actions/runs/12699806792/job/35401209406#step:15:7 ?

This only occurs in Async Demo example ( ie when AsyncWebServer is included ). The code compiles fine without AsyncWebServer.

mathieucarbou, 2025-01-09:
I reactivated the CI build with pio (was only there with arduino core): https://github.com/mathieucarbou/ESPAsyncWebServer/actions. All fine.

> @mathieucarbou Do you have any idea on why the following error is happening: https://github.com/ayushsharma82/NetWizard/actions/runs/12699806792/job/35401209406#step:15:7 ?
> 
> This only occurs in Async Demo example ( ie when AsyncWebServer is included ). The code compiles fine without AsyncWebServer.

Already seen that yes but do not remember. 
It is with arduino ci: is it possible that you need to include also ArduinoJson in the AsyncDemo example ?
I know Arduino IDE and CLI are doing shady things for library discovery by looking at includes.
Platformio at least has the lib_compat_mode and lib_ldf_mode that we can set.

In any case, do not forget to upgrade the dependencies ;-)

https://github.com/ayushsharma82/NetWizard/blob/main/.github/workflows/ci.yml#L78

ayushsharma82, 2025-01-09:
All deps are updated in CI of `dev` branch.

Similarly I'm facing this issue in arduino-cli test only, PIO CI is passing. Very odd behavior.

ayushsharma82, 2025-01-09:
Moving ArduinoJson include above AsyncJson.h of AsyncWebServer fixed it.

I noticed AsyncJson.h was detecting if ArduinoJson was included before it: https://github.com/mathieucarbou/ESPAsyncWebServer/blob/main/src/AsyncJson.h#L38

mathieucarbou, 2025-01-10:
> Moving ArduinoJson include above AsyncJson.h of AsyncWebServer fixed it.

Yes! Indeed it needs to be added before: I didn't notice looking at the code. This is because ESPAsyncWS acts like feature detection and activate the json code only of the ArduinoJson header is found in the include list.