mathieucarbou/ESPAsyncWebServer #192: [BUG]Crash with newer versions of ESP32 board package (Arduino IDE)

elaw01887, 2025-01-12:
Are you folks aware that the library seems to have a crashing issue with newer versions of the ESP32 board package (> 3.0.7) and at least some of the newer ESP boards (I'm using the ESP32S3)?

It's discussed here: [https://forum.arduino.cc/t/esp32-rebooting-loop-with-espasyncwebserver/1334275](url)

Here's a snippet of my code:
`    //ASYNC web server stuff:
    ws.onEvent(onAsycnWSEvent);  //onAsycnWSEvent is in WebServerStuff.ino
    server.addHandler(&ws);

    // Web Server Root URL
    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
        request->send(LittleFS, "/index.html", "text/html");
    });

    server.serveStatic("/", LittleFS, "/");

    // Start web server
    server.begin();
`
The crash happens when server.begin() is executed, and will happen even if I comment out the code above it.

I've tested with version 3.3.22 of the web server library and all newer versions and all seem to have the problem.

I'm using this board: [https://www.amazon.com/gp/product/B0D9W4Y3F3/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1](url), but the folks discussing the issue in the Arduino forum thread I think are using different ones.

I don't know how to capture a stack trace but will research it and post back if I can figure it out!

proddy, 2025-01-12:
Thought it was fixed in https://github.com/mathieucarbou/AsyncTCP/issues/27

mathieucarbou, 2025-01-12:
@elaw01887 : please make you you use the right dependencies before opening an issue. 
Your stack traces show that you are either not using the lastest versions of these librairies, or that you are not at all using ESPAsyncWS and AsyncTCP from my forks as described in his project readme.